FreeCAD/FreeCAD #8692: [Problem] Workbench dropdown is difficult for a new user to navigate

chennes, 2023-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.31802 (Git)
Build type: Release
Branch: master
Hash: 7352512f6c7e312728bcb64ccf9e7c890df983cd
Python 3.10.9, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Problem description

This was discussed at FreeCAD Day 2023:

In a new installation of FreeCAD, the standard workbenches list is arranged alphabetically and contains about twenty workbenches. This makes it difficult for a new user to discover the standard workflows in FreeCAD, and leads to clutter in the user interface.

We discussed a number of possible solutions (these are not mutually exclusive):
1) A first-run dialog that helps a new user select which workbenches they expect to use the most, and displays those more prominently in the user interface.
2) An easier-to-discover system for editing the workbenches menu (for example, an option at the bottom of that menu called "Edit...")
3) The ability to create submenus to either group those workbenches in some predefined way, or to allow less-used workbenches to be tucked away and not clutter the main list.
4) An option to auto-sort the list so that the most-recently-used workbench is at the top of the list.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-02-27:
I would combine No. 1 with a selector toolbar add-on style selector instead of the current genuinely annoying to use drop-down.

luzpaz, 2023-02-27:
@MisterMakerNL was working on a first run dialog.

ngocminhdao88, 2023-02-27:
I have an idea. How about a search box with fuzzy search. User types in few keywords and FreeCAD filters workbenches, which matched the search criteria. Maybe filter wb combobox contents when it focused and user typing is also a nice touch.

Intuos5, 2023-02-27:
> I have an idea. How about a search box with fuzzy search. User types in few keywords and FreeCAD filters workbenches, which matched the search criteria. Maybe filter wb combobox contents when it focused and user typing is also a nice touch.

That would inlo work well with an extensive command search, an extended version of Tripkus' launcher.

yorikvanhavre, 2023-02-28:
I would like to have a "first run" screen, maybe in the start page? Roughly based on https://wiki.freecad.org/Which_workbench_should_I_choose
Where can we see @MisterMakerNL 's work? I can't find it back...

MisterMakerNL, 2023-02-28:
I got 2 pages ready but the rest is still example code and so a bit off a mess.  That's why I didn't publish it too github.

Intuos5, 2023-02-28:
https://forum.freecad.org/viewtopic.php?t=71277&start=40
Is wgere the examples where posted

xtemp09, 2023-03-03:
I posted [an idea about that](https://forum.freecad.org/viewtopic.php?t=76424). Seems like people just ignored it. `QTableView` looks better than `QListView`. Addition of search with autocompletion also would be good.

luzpaz, 2023-03-14:
Related to #5772

PaddleStroke, 2023-03-14:
Is someone actively working on this ? @MisterMakerNL ? If not I can pick it up.

MisterMakerNL, 2023-03-14:
Getting started is live here:
https://github.com/FreeCAD/FreeCAD-Getting-Started
I'm not working on the workbench toolbar|
But any help is welcome, since I am learning while doing it all takes ages.

prokoudine, 2023-03-16:
Just my 0.02 euro.

All the suggested solutions have pros and cons. The Getting Started wizard, in particular, assumes that the user already knows what WBs the program has, what they do, and which ones the user needs, which is not necessarily the case, especially for new users.

A few thoughts.

When you are building a rocket, what do you need? Rocketry, Sketcher, maybe FEM and TechDraw. But definitely not Arch, or BIM, or Paths? When you are designing a house, you need Sketcher, Arch, BIM etc., but again, probably not FEM or Robot (unless you really, really need to). So there's context. 

Well, people have already figured out several ways to deal with context. One such way is workspaces that limit visible UI to just the features that belong to a particular context, with the rest tucked away but still fairly accessible. You can see this in multiple non-linear video editors like Kdenlive that come with a set of predefined workspaces yet give users the way to configure their own workspaces. For FreeCAD, that might mean having a context selector and dialog similar to Getting Started above for defining custom workspaces and managing all workspaces (factory defaults and custom ones), and a few other things maybe.

The tricky bit is figuring out the UI to choose context when starting a new project. It's easy when you add an extra step just for that. But the price is an extra step. Or you can do it similarly to what Inventor does: have a split button and multiple choices of what to create: a 2D sketch, a part, an assembly etc. And give 3rd party WB developers API to 1) plug into that split button menu, and 2) append a new workspace and set WBs visible by default (i.e. define context).

All this will become even easier after some internal cleanup, because it looks like users are not particularly fond of having two 2D sketching workbenches by default and would rather have just one. Which, again, removes the cognitive load.

prokoudine, 2023-03-16:
Just a quick and dirty prototype of a slightly different start window to illustrate my point, not polished at all:

https://user-images.githubusercontent.com/57467/225608730-8f6fb1cf-2b5d-4702-a1ff-79631e9d91e3.mp4

prokoudine, 2023-03-16:
Quick prototype of controlling what's visible on the start page:

https://user-images.githubusercontent.com/57467/225615619-dca5e1ca-a205-4803-8771-3b6e0244dd8c.mp4

yorikvanhavre, 2023-03-20:
The "start new project" tab is a really interesting idea...

One thing that bothers me here is that I think in order to successfully work with FreeCAD, a basic knowledge of workbenches is needed. And we cannot simply say "to do task XXX you need workbench YYY". So, one way or another, we need to teach new users about workbenches. I find it difficult to avoid that, and maybe even counter-productive to try (users may then lack of an important knowledge). It does not need to be complicated. Just so you know in what workbench to look for what kind of tool. You will necessarily need, sooner or later, to change workbenches.

I'd rather (my 2 cents here) not hide that fact to the users. Rather, indicate them where to start and explain the basics.

A "start new project" tab should come with explanations I think. I like that there, different "use cases" are identified. Maybe what we need is a short explanation right there?

sliptonic, 2023-03-21:
This issue is a bit too big.  It's good discussion but it's becoming about longer-term objectives that are harder to achieve.  We can make more progress in small steps.  Two very obvious quality of life improvements could be done in one or two days:

- 'ship' FreeCAD with some of the workbenches disabled. Robot, Reverse engineering, Web, Inspection, Test Framework at least
- Kill the abomination that is the workbench tab in the 'Customize' dialog. It's horrible.  Instead, add one more column of checkboxes to the workbench list in Preferences for the user to enable there.

PaddleStroke, 2023-03-24:
> Kill the abomination that is the workbench tab in the 'Customize' dialog. It's horrible. Instead, add one more column of checkboxes to the workbench list in Preferences for the user to enable there.

+1. Just done it, you can check it out here : https://github.com/FreeCAD/FreeCAD/pull/9026

Bryfour, 2023-05-25:
> ### Is there an existing issue for this?
> * [x]  I have searched the existing issues
>
> ### Version
> 0.21 (Development)
>
> ### Full version info
> ```shell
> OS: macOS 13.2
> Word size of FreeCAD: 64-bit [...]

1 and 4

Bryfour, 2023-05-25:
1 and 4, The new users can select what they will use the most, and nomal users may have an easier way of getting everything done. also, adding some new key combos would be nice.

Bryfour, 2023-05-25:
> I got 2 pages ready but the rest is still example code and so a bit off a mess. That's why I didn't publish it too github.
#######################
Someone on here told me to keep it within 100 lines because then its easier to implement. also people won't get mad unless its a big change, and ui is important to everyone

PaddleStroke, 2023-08-10:
As first run wizard seems to need some more time in the making, can we start cleaning the workbench selector by removing seldomly used wb? 
- Disable robot, inspection, reverse engineering.
- Disable OpenScad : This requires an external installation to work so there's no reason to be enable by default.
- Point workbench feels like it should be a toolbar of the Mesh workbench rather than an independant wb. Who's in charge of Point and Mesh to weight in?
- Disable Web wb by default. Not sure this is really useful for most users.

obelisk79, 2023-08-10:
agree with paddle's points 1-3. 

regarding the web wb, what is its purpose in FreeCAD? open the wiki? as a web backend for other functions etc? in any of these cases, it just shouldn't be included in the selector pulldown.

yorikvanhavre, 2023-08-10:
The idea of the web wb was basically to access websites where you canedownload resources (models, etc). if you click on a link to a step file, itteven allows you to open it directly in FreeCAD.

The start wb uses that web module indeed, but does not require the workbench. IMHO it could be hidden by default, but should still be there for power users to enable

PaddleStroke, 2023-08-11:
I'm not sure, this adds a mechanism and makes things more complex. Besides this might be more confusing.

What about tooltips for each workbench in the list with a short description ?
[image]

Intuos5, 2023-08-11:
Or, add a different button at the bottom "Other..." with the tooltip "Enable more workbenches in preferences" and you'll open the revamped workbench preferences.

qewer33, 2023-08-11:
I've made a mockup for new file templates for the start page based on this issue and some discussion that was held in the Discord server and forums.

[image]

Forum post: https://forum.freecad.org/viewtopic.php?t=80334
Github issue: https://github.com/FreeCAD/FreeCAD/issues/10165

This of course isn't a complete solution to the problem presented in this issue but it should be a great start. I am willing to take this on and implement this as my first PR to FreeCAD if people are in favor.

sliptonic, 2023-08-11:
This issue seems to have diverged.  The original problem was about the workbench dropdown.  The mockups are addressing a different problem with the start workbench.  Can we split this?

sliptonic, 2023-08-21:
I've created a project to gather issues together about workbench selector.
https://github.com/orgs/FreeCAD/projects/16/views/1

I've also created some issues to address individual workbenches.  There's a lot of good discussion in this issue so I'm not closing it yet but would encourage everyone to start discussing individual workbenches instead of the whole selector at once.

ant1ucem, 2023-09-12:
I usually use PartWorkbench, Part, Sketch, and Draft in my work. Occasionally I need other workbenches as well. And I think this is the same situation for 90% of users (correct me if I'm wrong)

The best solution for me would be not to search inside the dropdown for these 4 that I use all the time, but to navigate in them with one click. I  suggest placing direct jump buttons to the most popular workbenches before the dropdown. Or at least allow the user to place these buttons themselves through interface customization.

Should I create a separate detail issue for the suggestion?

FlachyJoe, 2023-09-12:
@ant1ucem you should look at the SelectorToolbar plugin: https://github.com/triplus/SelectorToolbar

ant1ucem, 2023-09-12:
@FlachyJoe well, thank you. But I don't asking for plugin. I suggest solution for the issue. 

If that solution already six years old, it means it is not that bad

sliptonic, 2023-09-12:
> The best solution for me would be not to search inside the dropdown for these 4 that I use all the time, but to navigate to them with one click. I suggest placing direct jump buttons to the most popular workbenches before the dropdown. Or at least allow the user to place these buttons themselves through interface customization.

It isn't mouse-accessible but there are already accelerator keys for this.  You can use the preferences to organize the order of the workbenches in the list and then (W, 1) will take you to the first.  (W, 2) to the second.  etc.

If you really want a button, you can write a one-line macro like 
Gui.activateWorkbench("DraftWorkbench") and link it to a toolbar button.

ant1ucem, 2023-09-12:
@sliptonic see issue adressing problem. As I already said personally I do not need advices

sliptonic, 2023-09-12:
@ant1ucem, this isn't about solving your problem.  The issue is about discussing alternative approaches to make the workbench selector more efficient.  You've suggested some options and other people are responding to your suggestions.  This isn't about offering you advice personally, it's about understanding what you're getting at and whether it is worth spending development effort to achieve.  

If I was going to offer you advice it would be that if someone tries to help you, the correct answer is, "thank you"

ant1ucem, 2023-09-12:
@sliptonic I'm confused. So are we discussing the selector problem or are you continuing to offer me advice?

qewer33, 2023-10-17:
There have recently been discussions on the Discord about a streamlined "tabbar" alternative to the WB selector.

The idea is, there is a tabbar that shows the currently loaded workbenches as tabs that the user can click and navigate. Unloaded workbenches do not show as tabs but there is a "plus" button at the end of tabbar that will show a dropdown with the unloaded workbenches so the user can click one and load it, it will be added to the tabbar.

This also has the potential to be integrated with the Start Page quick-start buttons. For example when the user creates a new file with the "Standard Part" button in the Start Page, only workbenches related to that specific workflow will be loaded.

Here is a mockup below to illustrate the idea better:

[image]

Would love to hear opinions from here. Also as a note, this can live together with the current WB selector for old time users that prefer it but it would ideally be the default.

adrianinsaval, 2023-10-17:
[SelectorToolbar](https://github.com/triplus/SelectorToolbar) works great for me and I wish it was the default experience:

[image]

showing icons only saves a lot of space, the ability to place it vertically also saves precious vertical space in the 3d view and of course the ability to choose which workbenches I have quick access to is great, better than relying on loaded workbenches IMO

yorikvanhavre, 2023-10-18:
drop-down, tab bar or toolbar are all valid forms and all have pros and cons, and if we made a poll we would probably get 33% of votes for each :)

Ideally, on a first time wizard, one should be asked: "How would you like your workbenches selector?"

I remember @sliptonic saying/writing somewhere that "first time wizard should be for those things where there is no possible consensus", maybe this would be a good case.

obelisk79, 2023-10-18:
I somewhat disagree with York. Here is why.

There has been very much discussion surrounding a first run dialog. One of the important principles for such an interface element is that it should only provide limited options that even users who are unfamiliar with FreeCAD could readily understand.

The current workbench selector is an awkward UI element that current experienced users are familiar with but exist as an enigma to newcomers. *If* a change is implemented, I would propose the old widget be available via a preference or parameter flag and be presented when a 'classic' theme is selected (through such a quick start dialog) so it does not fully alienate experienced users who desire things remain the way they are already comfortable with.

yorikvanhavre, 2023-10-18:
> I would propose the old widget be available via a preference or parameter flag

I'd be totally fine with that. The important is not to remove functionality that users have worked with for years.

sliptonic, 2023-10-18:
> I remember @sliptonic saying/writing somewhere that "first time wizard should be for those things where there is no possible consensus", maybe this would be a good case.

Be careful.  The important thing is no _possible_ consensus.   Units, language, mouse model are good examples.  Just throwing up our hands because some people don't like a change isn't a high enough bar yet.

We should be more risk-tolerant, especially in development.   Make the change and see what happens.  Either,
- People will largely like it (keep)
- People will largely dislike it (revert)
- People will become polarized (add option)

Just don't assume that people will be polarized before they've had a chance (been forced to) live with it a little while.

kadet1090, 2023-10-18:
>showing icons only saves a lot of space, the ability to place it vertically also saves precious vertical space in the 3d view and of course the ability to choose which workbenches I have quick access to is great, better than relying on loaded workbenches IMO

While I like the compactness of the icon only scheme, I don't think that it is viable solution. At least not as default. 

Workbenches do not have clear enough icon for new users to understand their meaning and purpose. Names are also not always very good. For example - "Path" should be called "CAM" or at least "Path (CAM)", the whole "Part" vs "Part Design" discussion etc. However, even in current state most of them are good enough for the users to understand the purpose so we should not hide them (by default).

### What should be our goals?
1. It MUST be clear to the new users how to switch workbenches and what workbenches (and by extension - abilities) are available
2. It SHOULD present most relevant workbenches to user in a clear way
3. It SHOULD make changing workbenches easy - ideally 1 click should be enough
4. It SHOULD allow user some degree of placement customization
5. It SHOULD work for larger workbenches counts

Icon only solution fails the first and most important point while addressing point 5. which is not as important IMHO as we should tend to have less loaded at once.

### How it can work?
As we discussed on the Design Working Group on Discord:
- Tab Bar for workbenches would be probably the most readable and most easy to access
- Tab Bar has limited space, it should display only loaded workbenches
- Workbenches that are not currently loaded should be available via "+" / "more" after tabs. After selecting workbench it is loaded and hence available in the Tab Bar.
- Auto Load feature in preferences will essential become "favorite" feature. Favorite workbenches will be always loaded, hence always visible.
- New File presets should load most common workbenches for given use case, that setting (workbenches to preload) should be stored in the Document. Documents that do not have this information stored should load some default set which covers most usecases.
- User should be able to reorder preloaded workbenches in a way that suits his or her usecase to keep their order stable. There is no need to reorient dynamically loaded workbenches.
- Tab Bar should be allowed on every toolbar space - top, bottom, left and right 
- Tab Bar could have optional "Compact Mode" which displays only icons - for users that know the software well enough
 
### Open Questions
- Should user be able to "unload" (hide) the workbench? 
- Should "loaded" state be per-document? i.e. should we allow different sets of workbenches per document in one "session" of FreeCAD

adrianinsaval, 2023-10-18:
> While I like the compactness of the icon only scheme, I don't think that it is viable solution. At least not as default.

I completely agree this should not be default, but could be a very useful feature to have for us with experience. Following the "tabs" paradigm we can mimic what browsers do:
[image]
pinned tabs are icon only while the rest is icon+text

>     * Tab Bar has limited space, it should display only loaded workbenches

and pinned WBs too please, preferably making showing loaded WBs optional. Then if somebody wants the old behavior they can just disable showing loaded WBs and not pin any WB.

>     * Auto Load feature in preferences will essential become "favorite" feature. Favorite workbenches will be always loaded, hence always visible.

I don't like this as a way of marking favorite WBs because this slows down startup, it would be better to have a way of marking favorites without forcing them to load at startup. See my proposal about pinning tabs.

>   There is no need to reorder dynamically loaded workbenches.

are we sure of this? does allowing this bring any more complexity or this just the whim of designers?

>     * There should be option in preferences to use "Combo Box style" switcher - which is the current one and takes less space.

I don't think this is really necessary, we can offer an option that functions effectively the same without requiring us to keep the old widget. Like I said early, the tab bar could show just the active WB and this would effectively work the same as the current selector.

>     * Should user be able to "unload" (hide) the workbench?

yes IMO

>     * Should "loaded" state be per-document? i.e. should we allow different sets of workbenches per document in one "session" of FreeCAD

This seems overkill, if done it should happen later if there is real desire for it.

kadet1090, 2023-10-18:
>pinned tabs are icon only while the rest is icon+text

I like this behavior in browsers, but I don't think that doing that both styles at once in that case would be beneficial. You either know workbenches by icons or not, I don't think that we need both styles at once displayed. It will increase complexity from code point of view, visual complexity and gains are minimal. But I will consult that with the rest of the group.

>are we sure of this? does allowing this bring any more complexity or this just the whim of designers?
From my understanding of how this can be done - yes it is more complex. You have static set of "favorite" workbenches, that are always placed in the same place. The problem is more complicated if you add dynamic loading to that because now you have multiple possible sets of workbenches to order. However, If there is simple solution that would work - I'm not against that, I simply thing that it is not as important as ordering favorites. I don't think however that newly loaded workbench should appear somewhere in the middle of the list, they should always appear on the end. Moving stuff in the UI is often considered bad practice.

>and pinned WBs too please, preferably making showing loaded WBs optional. Then if somebody wants the old behavior they can just disable showing loaded WBs and not pin any WB.

Yeah, I think that we can make "Favorite"/"Pinned" concept separated from "Loaded", it was too much of a shortcut probably.

>I don't think this is really necessary, we can offer an option that functions effectively the same without requiring us to keep the old widget. Like I said early, the tab bar could show just the active WB and this would effectively work the same as the current selector.

I meant there to keep something that works in the same way, it does not have to be old widget, simply old behavior.

closed 2024-04-01 by chennes