FreeCAD/FreeCAD #8693: [Issue] Integration tests should be separated from unit tests

berniev, 2023-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76133

### Version

0.21 (Development)

### Full version info

```shell
current
```

### Subproject(s) affected?

None

### Issue description

Integration tests involve more than an isolated piece of code. They are by their nature slow due to such things as dependencies, external accesses etc.

Unit tests on the other hand must operate on code isolated from dependencies etc (by stubbing, mocking etc) so they can run very fast.

Currently there is a set of tests (Metadata) that run very slowly and hence classed as integration tests. Unfortunately the design of the class does not lend itself to isolating its dependencies. But an integration test has some merit.

cmake target for unit tests is currently Tests_run.
I propose a separate cmake target for slow tests, perhaps "slow_tests", so they can be run separately.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-28:
Is the 'metadata' mentioned here related to chennes's Addon Manager tests ?

adrianinsaval, 2023-02-28:
if the test is important, is there a situation where it makes sense to skip it?

berniev, 2023-02-28:
@adrianinsaval Slow tests should simply NOT be part of normal unit tests because unit tests must run fast so they can be run frequently during development. There can be thousands of tests and the idea is the whole lot runs in seconds. The faster the tests, the more often they are likely to be run, and problems detected long before a project build.

**I appreciate this is different to what some here insist unit testing is.** 

Having said that the unit testing framework (gtest) is quite capable of running slow (integration etc) tests. We just don't want slow tests to spoil unit testing.

Sooner rather than later CI should include unit tests. CI has less time pressure so conceivably could run both unit tests and slow tests.

StefanBruens, 2023-08-08:
Typical approach for slow test is labeling/tagging of tests. You can then by default skip any tests which are labeled as slow (or require external network access, which is not available on isolated build hosts, etc), and optionally run them.

ctest:
```
  -L <regex>, --label-regex <regex>
                               = Run tests with labels matching regular
                                 expression.  With multiple -L, run tests
                                 where each regular expression matches at
                                 least one label.
```

luzpaz, 2023-08-16:
CC @chennes (I think the metadata tests refer to your code in this discussion)

chennes, 2023-08-17:
I'm fine with either proposal. If we are going to have a distinction between "slow" and "not slow" tests we should provide some guidance to test writers as to what constitutes "slow", e.g. "a test that takes longer to execute than to compile", "a test that takes more than 10ms on the CI hardware", "a test that loads an external shared library" etc. ... whatever is appropriate. If the distinction is more a matter of principle then someone (not me!) has to write out how one decides where to put what test.

berniev, 2023-08-17:
I think it is best to stick to industry standard norms.

`Unit tests` and  `Integration tests` serve different purposes.

`Unit tests` test an `isolated` unit of code and are inherently `fast` as a result. Dependencies are replaced by mocks, stubs, etc so the test is laser-like focused only on the unit under test. This enables many many unit tests to be run very very quickly and so the tests can and should be run very frequently.

`Integration tests` on the other hand test a `number of code units simultaneously` -- due to dependencies being included with the code under test. This results in lesser performance and brittle tests (due to the number of code units that could later undergo change and break the test). Due to lower performance, Integration tests are not able to be run as frequently as unit tests.

The impact of mixing unit and integration tests may not be apparent right now, but compared to the size of the codebase the number of tests so far is insignificant but as the number of tests rises, so will the performance hit.

Currently FreeCAD is still learning C++ testing. Many tests purport to be "unit test" but are actually integration tests, probably because they are much easier than getting legacy code under unit test. It is hoped that over time the codebase will modernise and unit testing become easier.

Langr pp 279-280 is well worth a look.

berniev, 2023-08-17:
> a test that takes longer to execute than to compile

You have mentioned this repeatedly but havn't explained the conjunction. As discussed, a unit test will run well under sub millisecond. Of course compiling takes longer, but so what? An integration test can take much longer (I have seen some here take 200mS+). If a test includes lots of stuff that includes lots ... likely compile will take longer, no?

chennes, 2023-08-17:
My point is that when testing C++ one always has to compile, so it sets a floor for how fast any test can be during development. 

> Dependencies are replaced by mocks, stubs, etc so the test is laser-like focused only on the unit under test.

There is _not_ widespread agreement that mocking/faking/stubbing everything is beneficial. I spoke with @JohnAD about it this past week, and there is plenty of argument to be made that mocking can be actively harmful.

My copy of Langr is digital, what section are you citing above?

berniev, 2023-08-17:
>  there is plenty of argument to be made that mocking can be actively harmful.

That is an empty, unsupported statement of no substance. Five minutes of search will give you zillions of sources of information explaining the theory and practise.

> My point is that when testing C++ one always has to compile, so it sets a floor for how fast any test can be during development.

I don't know what you are doing wrong, but after initial run compile should be no more than a few seconds after a code change. Running tests of course is nearly instantaneous. If your test is an integration test with a large include tree (which a number of your tests are) on code with lots of dependencies, things are going to blow out. I regularly run tests after every couple of lines changed. 

Once code is isolated from the rest of the world, there is very little to compile!

Unit testing is not just about the tests, but a reflection of how code is written. It makes you think about the structure of code, not just piling more and more into massive classes/files with lots of dependencies. Legacy spaghetti code like FreeCAD is largely the antithesis of (unit) testable code, but one hopes that will change over time as skills improve. 

Re Langr: S10.3

I'm sure the penny will drop for you one day.

chennes, 2023-08-17:
We've gotten pretty far afield of the original discussion issue report: you've proposed that we have a separate target for slow tests. If we want to have such a thing, we have to tell people what "slow" means. What do you propose?

berniev, 2023-08-17:
The split I propose is between unit tests and integration tests (see previous posts). Speed is a by product

chennes, 2023-08-17:
I doubt that we'll be able to convince many developers to write "unit tests" at all, and we'll just end up with everything in integration tests. Since the current target is called "tests" and is not particularly slow (as you point out, it compiles in a few seconds and runs in a few milliseconds), it probably makes the most sense to close this now and revisit if we ever get to a point where there are enough tests for it to matter. Any other maintainers have opinions either way? @wwmayer you've written the most tests, what's your inclination here?

berniev, 2023-08-17:
>  as you point out, it compiles in a few seconds and runs in a few milliseconds

**Please do not misquote my words**. I'm talking about my unit tests more than 100 of which don't add up to 1mS.

The current tests (mostly integration) take up to hundreds of mS to run, and there are very few of them. Its only going to get worse.

wwmayer, 2023-08-17:
> Currently there is a set of tests (Metadata) that run very slowly and hence classed as integration tests.

Do you have measured the time it takes? If yes how long does it take? Currently the Metadata.cpp includes the test cases below with the measured time (they are part of Tests_run):
* ContactTest (total time: 0ms)
* LicenseTest (total time: 0ms)
* UrlTest (total time: 0ms)
* VersionTest (total time: 0ms)
* MetadataTest (total time: 0ms)

Currently Tests_run includes 383 tests (32 test suites) and their total time is ~120 ms. The "bottlenecks" are ReaderTest (4 ms, 11 tests), ComplexGeoDataTest (20 ms, 24 tests), ElementMapTest (13 ms, 16 tests), all other tests (0 ms). I guess the vast majority of the 120 ms is needed to load and unload the 3rd party shared libraries.

So, currently I don't see a problem with any of the test suites we have in Tests_run.

> I propose a separate cmake target for slow tests, perhaps "slow_tests", so they can be run separately.

In the future we will have some tests that require an initialized Python interpreter or that require FreeCAD's type system to be set-up. For them I guess it makes sense to have a separate target since therefore FreeCAD's application framework must be started and this might be slow.

> I doubt that we'll be able to convince many developers to write "unit tests" at all

At some point we have to force them. If a PR is made that adds new functions or modifies existing functions then a unit tests must be mandatory.

chennes, 2023-08-17:
> At some point we have to force them. If a PR is made that adds new functions or modifies existing functions then a unit tests must be mandatory.

I think we can say that *tests* are mandatory, but unless/until we make it much easier to write true "unit tests" I think we're just going to end up with "integration tests" -- and frankly, I think that's fine. The internet is full of people arguing about what a "unit test" is. If we just call them "tests" we can stay out of that fight.

berniev, 2023-08-18:
>  tests are mandatory

Not at the moment..

> until we make it much easier to write true "unit tests"

"We" are never going to make it easier! "We" are just going to have to learn some modern  programming principles.

> The internet is full of people arguing about what a "unit test" is

Its really disappointing to see comments like this designed to mislead those who don't know better.

JohnAD, 2023-08-18:
I see that my name has been rolled into this discussion; so I'll leave a few comments.

(A bit of disclaimer: most of my experience of the last 30 years is based on corporate shops rather than open source. This is especially true of TDD as it gained stronger prominence in the 2005 to 2018 era.)

In general, segregating slower tests from faster tests sounds like a reasonable idea. We can start with a simple definition of "slow" and change that over time as opinions change.

Regarding the definition of "integration tests", the original definitions were aimed at system dependencies; especially stateful ones. For example, calling an external API or having a test include database access. Later, this was expanded to mean "complex internal systems" also, but the line for what is complex and non-complex is judgement call.

Some folks also simply call any test that "calls other code already covered by tests" an integration test. In some environments, such as Javascript frameworks, this is sometimes a good definition. In those environments you want the tests running while you are typing in dynamic code. The tests should be instantaneous (or nearly so). Then, later, you run more comprehensive tests manually and before commits and during CI/CD. If the class or code you are working calls systems already tested, you just get to the habit of running a shell on the side to run the "other tests" quickly.

This is all different discussion than "white box" test; where the core section (or all) of the application has to be launched to run the tests; even if they are not "integration tests" by the older definitions; "white box" tests are almost **never fast**.

I bring this up because the while the ComplexGeoData tests we wrote are not white box, they came awfully damn close. And, in this next phase of TopoNaming changes MOST of the tests will be white box. The tests will literally build a document with example objects and invoke workbench operations. Many hundreds of interconnected classes will be involved.

So, yes, having a `slower_tests` cmake target could be very useful :smile: .

Now, as to automatic (reflective) mocking libraries (as opposed to manually setting-up mocks (stats) or fakes (no stats)): I do indeed believe you should be cautious using them. I've seen too many tests that "prove" that broken code is still working leading to weird and long-running code hunts. The very reflective code-generation that makes them so easy to use also makes it easy to write bad tests. Some companies have outright banned them after too many problems; though I think that is perhaps a bit extreme. They are really handy sometimes.

JohnAD, 2023-08-18:
Personally, I tend to separate "unit test" from "integration test" based on the goal; which is the really old definition. An integration test (or I&T) covers the integration of multiple units. But a unit test limits coverage the internal business logic or operation of that one unit. So a single unit, such as a class, can have both integration testing and unit testing. And the unit testing need not have mocks/fakes for unit dependencies if those dependencies also have their own unconnected units tests and the unit is not stateful. If it is stateful (even when internal) then a make/fake is absolutely required.  But that is just my two cents.

I do know of Langr's work on this also, at least his Java books. I see that he has a C++ book; that sounds like a useful read.

berniev, 2023-08-18:
Thank you @JohnAD for stepping in.

TL;DR: Your understanding of the issues appears to be essentially the same as mine :clap:. However, different backgrounds are impacting communication and understanding. There are testing newbies here.

> Langr's work

"Modern C++ Programming with Test-Driven Development (Code Better, Sleep Better)" Jeff Langr 2013. Highly recommended.

> "white box", "black-box"

White-box testing is a concept of testing that is similar to that used here, but seeks to apply a different abstraction. Wikipedia is revealing: `White-box testing can be applied at the unit, integration and system levels of the software testing process.`

***I have twenty one C++ and related textbooks (learning!). **NONE of them has white-box testing in their index. Indeed, I had heard of it, but have never seen it referenced in any C++ book, tutorial, YT, lecture etc.  Given this, I suggest we are all better off using the current industry standard terminology.***

> Now, as to automatic (reflective) mocking libraries

I think you are referring to what is found in Java etc. Languages that have reflection often have automated test generaltion nowadays. Poor old C++ is missing out though. Perhaps your distrust of these was a trigger for the spray by another against mocking in an earlier post.

> This is especially true of TDD as it gained stronger prominence in the 2005 to 2018 era

Not sure what this means in the context of your 30 years in corp shops? Does this mean you missed out on TDD?

> such as Javascript frameworks

Interesting paragraph, but not helpful here.

> ComplexGeoData tests ... Many hundreds of interconnected classes will be involved

By any definition this is integration testing!

As an aside, there are tests of StringHasher (can't remember exactly) that fire up python. Big and slow! But when you look at the class under test it turns out there is just a couple of python related functions there, which the tests have duly mirrored. I would argue this is a case of asking wtf this class is doing depending on python? Was it REALLY necessary, or just architecturally lazy?

**SUMMARY: Integration v Unit**

I believe we agree a definition of `Integration Testing` is tests involving dependencies that are more than simple library (inherently trusted, super well tested) code (esp std library etc) and primarily (or accidentally) target the relationships between various units of code. Most user code never achieves the same degree of certainty. These will tend to be slow, but that is a side effect. They will be fragile, as their validity is dependent on nothing in the dependency chain changing.

In contrast, `Unit Testing` is test on a unit of code with untrusted or complex dependencies eliminated. They happen to be fast. Their validity is unaffected by changes to any other units of user code.

closed 2025-01-14 by github-actions[bot]