FreeCAD/FreeCAD #8700: [Issue] "External geometry" on shapebind ellipse arc recreates the whole ellipse

CharlesMir, 2023-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76408

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

To recreate the bug : 

1) Create a sketch with an elliptic arc in it 
2) Create a new body (partdesign) by "shapebinding" the elliptic arc from the previous sketch 
3) Create a new sketch inside this body
4) Use the tool "External geometry" on the shapebind 
5) BUG : The whole ellipse is recreated and not just the arc. 

It seems to be the only type of arc affected. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fra589, 2023-07-10:
Hi all, 

There is another forum discussion for what it seams to me the same bug, but without "new body (partdesign) by "shapebinding"":
https://forum.freecad.org/viewtopic.php?t=79517

Externals elliptical arcs and externals circular arcs are affected by this bug.

first proposal patch (but not complete) here:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Sketcher/App/SketchObject.cpp#L7629-L7641

@++;
Gauthier.

luzpaz, 2023-09-24:
@fra589 the link to the patch is not valid AFAICT

fra589, 2023-09-25:
> @fra589 the link to the patch is not valid AFAICT

The patch link from @Syres916 (proposed in the forum discussion https://forum.freecad.org/viewtopic.php?t=79517) was good in July 07... But not today, probably due to the many updates in the concerned file : FreeCAD/src/Mod/Sketcher/App/SketchObject.cpp

I'm sorry, but I tested it with partial success in July, but I lost it after testing. I seem to remember that the change proposed in the patch was in the `SketchObject::rebuildExternalGeometry()` function.

Perhaps @Syres916 will remember...

Syres916, 2023-09-25:
The reason the link is no longer valid is I'd got too many branches as it is/was and I must have cleared it out due to the lack of recent feedback, sorry but these issues just seem to drag on way too long for the way I'm used to work. I raised the fact that there are somewhere in the region of 150 Sketcher current Issues on the forum when the maintainer returned and there needs to be a priority order but there was no reply to that.

luzpaz, 2023-09-29:
@Syres916 thanks for all your hard work. I really want to see it addressed seriously as well. 

Just a tip, if I may? Perhaps label your branch in a way that you know what issue it's targeting ?

luzpaz, 2023-12-17:
@PaddleStroke can you take a look at this when you have a moment ?