FreeCAD/FreeCAD #8702: Create arrow slot is not shown adequately in 3D view

todork, 2023-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: Bulgarian/Bulgaria (bg_BG)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Hi,

I am trying to create narrow slot with rounded corners

[image]

in the 3D view I see

[image]

Is this a bug or I have done something wrong.

If the slot is created wider the problem disappears.

Regards,
Todor 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-28:
Can you please attach .FCStd file (zip it and drag/drop it in to this ticket, thx)

dnt2git, 2023-02-28:
Hi, More of a guess here, but it looks like the sketch isn't fully constrained. Specifically, the radius/diameter of the circles doesn't seem to have been set. One would be able to tell for sure when .FCStd is attached, as luzpaz said. The lines would appear green when the sketch is fully constrained.

todork, 2023-02-28:
Hi,
Thanks for the prompt reply.

Yes, here may be the example is not fully restricted, but I have checked different variants with the same  unsuccess.

Pls. find attached the .FCStd file
[BackCover.zip](https://github.com/FreeCAD/FreeCAD/files/10853451/BackCover.zip)

Regards,
Todor

todork, 2023-02-28:
Thanks again for the replies and especially to @dnt2y!!!!

After fully constraining the sketch the problem disappeared.

Regards,
Todor

closed 2023-02-28 by todork