FreeCAD/FreeCAD #8703: [Problem] topologicpy cannot be imported under 0.21. Worked perfectly under 0.20

wassimj, 2023-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
I can no longer import topologicpy under FreeCAD 0.21. It works perfectly under FreeCAD 0.20. Now I get the error:
>>> import topologicpy
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\site-packages\topologicpy\__init__.py", line 19, in <module>
    import topologic
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ImportError: DLL load failed while importing topologic: The specified procedure could not be found.
```

### Subproject(s) affected?

None

### Problem description

Many would like to integrate topologic with FreeCAD since both are based on OCCT. We need to ensure that topologic continues to be compatible with FreeCAD. With the new version, this is no longer possible. Please either fix or give me instructions on what I need to change in topologic to continue its compatibility with FreeCAD.

### Anything else?

topologicpy is available via pip install topologicpy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2023-02-28:
Discussion on test on Fedora 36 -

https://community.osarch.org/discussion/131/topologic-redefining-bim-through-spatial-topology-information-and-grammars#latest

https://community.osarch.org/discussion/1204/topologicpy-an-independent-python-api-for-topologic#latest

    "Hmmm, searched on my Fedora 36 and seems found it under .local/lib/python3.10/site-packages ...

    import sys
    sys.path.append("/home/paullee/.local/lib/python3.10/site-packages")
    import topologicpy

    Traceback (most recent call last):
    File "", line 1, in
    File "/home/paullee/Downloads/FreeCAD_weekly-builds-31391-2022-12-22-conda-Linux-x86_64-py310/squashfs-root/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
    File "/home/paullee/.local/lib/python3.10/site-packages/topologicpy/__init__.py", line 16, in
    import topologic
    File "/home/paullee/Downloads/FreeCAD_weekly-builds-31391-2022-12-22-conda-Linux-x86_64-py310/squashfs-root/usr/lib/python3.10> /site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
    ImportError: /home/paullee/.local/lib/python3.10/site-packages/topologicpy/bin/linux/topologic/./libTopologicCore.so.0: undefined symbol: _ZN24BRepBuilderAPI_MakeShape5BuildEv

    "

chennes, 2023-03-01:
What version of Python was your copy of FreeCAD compiled against?

paullee0, 2023-03-01:
Thanks, my version:-

FreeCAD_weekly-builds-31625-2023-01-19-conda-Linux-x86_64-py310.AppImage
on 
Fedora36

    OS: Fedora Linux 36 (Workstation Edition) (GNOME/gnome)
    Word size of FreeCAD: 64-bit
    Version: 0.21.0.31625 (Git)
    Build type: Release
    Branch: master
    Hash: 72df7997700062b6a02f2f208e6291f64073fb88
    Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
    Locale: English/United States (en_US)
    Installed mods: 
      * dodo 1.0.0
      * DynamicData 2.46.0
      * ArchTextures
      * BOLTSFC
      * FreeCAD-BlenderBIM 2022.10.20
      * BIM 2021.12.0
      * 3rd link to Paul
      * 3rd link to SketchArchPlus
      * lattice2 1.0.0

luzpaz, 2023-03-01:
@wassimj please add the full [About](https://wiki.freecad.org/About) info in to the original ticket

chennes, 2023-03-01:
Are you able to use it *outside* of FreeCAD? I get an error:
```
  File "main.py", line 1, in <module>
    import topologicpy
  File "/Applications/CLion.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "/Users/chennes/Documents/pytop_test_delete/.venv/lib/python3.10/site-packages/topologicpy/__init__.py", line 19, in <module>
    import topologic
  File "/Applications/CLion.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'topologic'
```

wassimj, 2023-03-01:
SOLVED!
It turns out, I had my old Topologic Workbench in my C:\Users\[username]\Appdata\Roaming\FreeCAD\Mod\ folder which was incompatible with FreeCAD 0.21.0 This was being loaded first before the newer topologic included in topologicpy.
It is now working fine on Windows.

[image]

luzpaz, 2023-03-01:
PEBKAC

closed 2023-03-01 by luzpaz

paullee0, 2023-03-01:
> SOLVED! It turns out, I had my old Topologic Workbench in my C:\Users[username]\Appdata\Roaming\FreeCAD\Mod\ folder which was incompatible with FreeCAD 0.21.0 This was being loaded first before the newer topologic included in topologicpy. It is now working fine on Windows.
> 
> [image]

Congratulation!  Hope RPM (or whatever) would be available on Linux (Fedora) so I can test