FreeCAD/FreeCAD #8706: [Feature Request]  Subtract the outside of the Sketch on PD

leoheck, 2023-03-01:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

This idea was presented here https://forum.freecad.org/viewtopic.php?p=664277#p664277

### Subproject(s) affected?

PartDesign

### Idea description

Instead of pocketing (or any other feature that removes material) with the Sketch, this setting would do that for the area outside o it.

Similarly, this could be extended to other features that add mass like Pad, except that it cannot work if it is applied to the first feature. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-01:
How would it work for a pad? It would generate infinite shapes, this only makes sense for substractive features IMO.

FlachyJoe, 2023-03-01:
> How would it work for a pad? It would generate infinite shapes, this only makes sense for substractive features IMO.

The shape could be limited by existent face borders.
[image]

adrianinsaval, 2023-03-01:
I don't see how the software is supposed to identify such boundaries, and I've never seen such an option in other software, I consider that out of scope for now. Is there even an use case for this?

FlachyJoe, 2023-03-01:
> I don't see how the software is supposed to identify such boundaries,

By projection of the nearest surface to the sketch plane.

> I've never seen such an option in other software

 I've seen it in Creo Parametric

> Is there even an use case for this?

Can't say… :slightly_smiling_face:

adrianinsaval, 2023-03-01:
If someone legitimately wants that, please create a separate issue as it works fundamentally different and I want to focus on subtractive features here.

leoheck, 2023-03-01:
> > How would it work for a pad? It would generate infinite shapes, this only makes sense for substractive features IMO.
> 
> The shape could be limited by existent face borders. [image]

Exactly like that.

adrianinsaval, 2023-03-01:
to focus the discussion here I'll quote myself from the forum:

> I think it's possible to tell occt to invert the face and do a pad from that without needing to calculate a contour around that, if not I'm thinking about taking the major diagonal of the bounding box of the base feature and making a big circle with that as radius.

and link to an old discussion/attempt at this: https://forum.freecad.org/viewtopic.php?t=61265

adrianinsaval, 2023-03-01:
> I think we should just disallow any pattern/mirroring on subtractive features when the "remove outside" option is used.

a response from the previous discussion highlighted in the forum, I can think of 3 ways to deal with the patterning problem:

1. Forbid patterns from features of this type, this complicates the implementation more as we would need special logic in transform feature to detect and reject these. Plus I believe there are situations where patterning this can make sense, for example a linear pattern along the extrusion axis. It was true for the case when using boolean common that patterns from it wouldn't make sense.
2. Don't do anything, keep transformation behavior as is and let the user deal with the results, can give expect results or not depending on the situation
3. Add special logic in transformations to deal with the addsubshape of these features, still not clear what operations would be required to get good results.

luzpaz, 2023-11-14:
@bgbsww any interest ?

bgbsww, 2023-11-14:
@luzpaz  Is this any different than the #11389 proposed solution to #5991?  I think I just implemented something very close to this.

jbaehr, 2023-11-15:
I think "remove outside or a sketch" is not always the same as "use common instead of cut". The use of "common" works only for 3D primitives like a cube or sphere, or when you "pocket through all".

But as the profile-based operations are not always done "though-all", they need a different approach to "remove outside". [This forum post](https://forum.freecad.org/viewtopic.php?p=664581#p664794) gives some examples. The difference arises when you make a pocket only to a certain depth. When using "common", also the remaining material along the profile's normal would be gone. But for "remove outside" I would expect it to remain. The "outside" must only be interpreted as "outside of my profile", i.e. in 2D. But common removes also the "outside in 3D".

leoheck, 2023-11-16:
I didn't follow the post, don't have much time and I am not sure about the details of the implementation. But I would try to find a polygon that can fit the current view of the object and also the sketch to make the subtraction.

FlachyJoe, 2023-11-16:
Maybe the face can be reversed in the inner/outer sens before make the prism. OCC can deal with infinite solid.

bgbsww, 2023-11-16:
Thank you, I get it now, I will take a look.

bgbsww, 2023-11-17:
Added to PR #11389.  Works, improvement suggestions are welcome.

closed 2026-09-03 by kadet1090