FreeCAD/FreeCAD #8707: [Problem] Draft_Dimension crash when picked point is out of working plane

FlachyJoe, 2023-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.15162 (Git shallow at 099de1a )
Build type: Unknown
Branch: master
Hash: c437b79adabb6a91e8fb6dc46729cb720ba16a5f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * sheetmetal 0.2.49
  * fc_nodes
  * A2plus 0.4.56
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * mooc-workbench
  * Render 2022.1.0
  * Plot 2022.4.17
  * lattice2 1.0.0
  * fcpd-workbench 0.2.0
  * Silk
  * Wood-Frame
  * Assembly4 0.11.12
  * fasteners 0.3.41
  * BIM 2021.12.0
  * Nodes 0.1.36
  * MeshRemodel 1.8919.0
  * fcgear 1.0.0
  * SelectorToolbar
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

1. Create a Part_Cube with default size
2. Set Draft working plane as Top
3. Start a new Draft_Dimension with snapping at segment ends
4. pick two point **at the top** of the cube
5. when clicking the third point to place the dimension, FreeCAD crash

### Anything else?

```
Pick a face, 3 vertices or a WP Proxy to define the drawing plane
----------------
GuiCommand: Dimension
Pick first point
----------------
Linear dimension
p1: Vector (10.0, 0.0, 10.0)
p2: Vector (10.0, 10.0, 10.0)
dim_line: Vector (16.153026254519233, 5.318159495255664, 0.0)
<Exception> Rotation.cpp(226): setValue(matrix): Could not determine the rotation.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bf90) [0x7f5d4fe5af90]
#1  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyObject_IsTrue+0x1f) [0x7f5d4f3cb86f]
#2  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x323e) [0x7f5d4f30843e]
#3  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab4a) [0x7f5d4f46ab4a]
#4  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x17f594) [0x7f5d4f37f594]
#5  0x7f5d5243d668 in Base::pyCall(_object*, _object*) from /usr/local/lib/libFreeCADGui.so+0x2a
#6  0x7f5d50bcc5c3 in App::FeaturePythonImp::onChanged(App::Property const*) from /usr/local/lib/libFreeCADApp.so+0x223
#7  0x7f5d50bd154f in App::FeaturePythonT<App::DocumentObject>::onChanged(App::Property const*) from /usr/local/lib/libFreeCADApp.so+0x83
#8  0x7f5d50c65ae5 in App::Property::hasSetValue() from /usr/local/lib/libFreeCADApp.so+0x59
#9  0x7f5d50c8112f in App::PropertyVector::setValue(Base::Vector3<double> const&) from /usr/local/lib/libFreeCADApp.so+0x61
#10  0x7f5d50c812ab in App::PropertyVector::setPyObject(_object*) from /usr/local/lib/libFreeCADApp.so+0x6f
#11  0x7f5d50c76fe7 in App::PropertyContainerPy::setCustomAttributes(char const*, _object*) from /usr/local/lib/libFreeCADApp.so+0x26b
#12  0x7f5d50c72da0 in App::PropertyContainerPy::_setattr(char const*, _object*) from /usr/local/lib/libFreeCADApp.so+0x2c
#13  0x7f5d50b1dc88 in App::ExtensionContainerPy::_setattr(char const*, _object*) from /usr/local/lib/libFreeCADApp.so+0x60
#14  0x7f5d50b3ce48 in App::DocumentObjectPy::_setattr(char const*, _object*) from /usr/local/lib/libFreeCADApp.so+0x60
#15  0x7f5d50bd27c0 in App::FeaturePythonPyT<App::DocumentObjectPy>::_setattr(char const*, _object*) from /usr/local/lib/libFreeCADApp.so+0x8e
#16  0x7f5d50bd235d in App::FeaturePythonPyT<App::DocumentObjectPy>::__setattro(_object*, _object*, _object*) from /usr/local/lib/libFreeCADApp.so+0xc1
#17  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyObject_SetAttr+0x50) [0x7f5d4f3cfb30]
#18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x814) [0x7f5d4f305a14]
#19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd5c) [0x7f5d4f5ccd5c]
#20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f5d4f468891]
#21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7339) [0x7f5d4f4b7339]
#22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f5d4f4b84eb]
#23  0x7f5d52387124 in Gui::Application::sDoCommand(_object*, _object*) from /usr/local/lib/libFreeCADGui.so+0x102
#24  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7f5d4f3c9ba1]
#25  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f5d4f37be8d]
#26  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f5d4f309c63]
#27  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab4a) [0x7f5d4f46ab4a]
#28  0x7f5d4f1e01d1 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /lib/x86_64-linux-gnu/libpyside2.cpython-311-x86_64-linux-gnu.so.5.15+0x91
#29  /lib/x86_64-linux-gnu/libpyside2.cpython-311-x86_64-linux-gnu.so.5.15(+0x17d1e) [0x7f5d4f1e2d1e]
#30  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8cba) [0x7f5d4dee8cba]
#31  0x7f5d4deecd6a in QTimer::timeout(QTimer::QPrivateSignal) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x3a
#32  /usr/lib/python3/dist-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(+0x2e68f7) [0x7f5d344e68f7]
#33  0x7f5d4dedd50d in QObject::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9d
#34  /usr/lib/python3/dist-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(+0x2e6487) [0x7f5d344e6487]
#35  0x7f5d4eb62fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#36  0x7f5d5245b99e in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib/libFreeCADGui.so+0x104
#37  0x7f5d4deb16f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#38  0x7f5d4df08c31 in QTimerInfoList::activateTimers() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x391
#39  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x3094c4) [0x7f5d4df094c4]
#40  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f5d4c11e7a9]
#41  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f5d4c11ea38]
#42  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f5d4c11eacc]
#43  0x7f5d4df09836 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#44  0x7f5d4deb017b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#45  0x7f5d4deb82d6 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#46  0x7f5d523201c0 in Gui::Application::runApplication() from /usr/local/lib/libFreeCADGui.so+0x2910
#47  FreeCAD(+0x133ff) [0x556f2536b3ff]
#48  /lib/x86_64-linux-gnu/libc.so.6(+0x2718a) [0x7f5d4fe4618a]
#49  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f5d4fe46245]
#50  FreeCAD(+0x12621) [0x556f2536a621]

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-01:
No crash here using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32146 (Git)
Build type: Release
Branch: master
Hash: 04d494cb75e180a4805feedc6d9c96e2ec306f97
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.6
  * PieMenu 1.2.4
  * Plot 2022.4.17
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.54
```

Here is my Python Console output for you to check:

```
Gui.activateWorkbench("PartWorkbench")
### End command Std_Workbench
### Begin command Part_Box
App.ActiveDocument.addObject("Part::Box","Box")
App.ActiveDocument.ActiveObject.Label = "Cube"
# Object created at document root.
App.ActiveDocument.recompute()
Gui.SendMsgToActiveView("ViewFit")
### End command Part_Box
Gui.runCommand('Std_DlgPreferences',0)
### Begin command Std_Workbench
Gui.activateWorkbench("DraftWorkbench")
### End command Std_Workbench
FreeCAD.DraftWorkingPlane.alignToPointAndAxis(FreeCAD.Vector(0.0, 0.0, 0.0), FreeCAD.Vector(0, 0, 1), 0.0)
FreeCADGui.Snapper.setGrid()
Gui.runCommand('Draft_Dimension',0)
# Gui.Selection.addSelection('Unnamed','Box','Edge12',0.243383,10,10)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed','Box','Vertex7',10,10,10)
import Draft
_dim_ = Draft.make_linear_dimension(FreeCAD.Vector(0.0, 10.0, 0.0), FreeCAD.Vector(10.0, 10.0, 0.0), dim_line=FreeCAD.Vector(4.498247146606445, 11.552892684936523, 0.0))
# Gui.Selection.addSelection('Unnamed','Dimension')
Draft.autogroup(_dim_)
FreeCAD.ActiveDocument.recompute()
```

FlachyJoe, 2023-03-01:
> _dim_ = Draft.make_linear_dimension(FreeCAD.Vector(0.0, 10.0, 0.0), FreeCAD.Vector(10.0, 10.0, 0.0), dim_line=FreeCAD.Vector(4.498247146606445, 11.552892684936523, 0.0))

Your picked points are on z=0 plane. In my case the two first are z=10 and the last z=0

FlachyJoe, 2023-03-01:
I've compiled the last git and issue is no more here.

closed 2023-03-01 by FlachyJoe