FreeCAD/FreeCAD #8708: [Problem] Misleading conflict shortcuts error message and inability to change shortcuts

Fat-Zer, 2023-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Gentoo Linux (Trinity/trinity)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32083 (Git)
Build type: Debug
Branch: master
Hash: 85ad88d101fb2f5eccf62e096f65a79f348d5f65
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.7.0
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * slic3r-tools
  * BIM 2021.12.0
  * ExtremeProDark 2.7.0
  * Glass (Disabled)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When some workbench has conflicting shortcuts an error message is shown that says:

```
The key sequence 'W, A' is ambiguous. Use 'Configure Keyboard Shortcuts'
from the 'Settings' menu to solve the ambiguity.
No action will be triggered.
```

But there is neither a menu called 'Settings' nor a 'Configure Keyboard Shortcuts' entry. At least I couldn't found such. There is a tab in `Tools->Customize->Keyboard`, but (1) it suggests that one shortcut should be prioritized over another in case of ambiguity and (2) the field 'Current shortcut' as well as 'Assign' button are appear to be disabled, so I see no way to change a shortcut.

So to recap the issues are:

1. A misleading (probably obsolete) error message
2. (Probably) Incorrect conflicting shortcut handling
3. GUI doesn't provide an ability to modify a shortcut

----
 
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-03:
Hi @Fat-Zer nice to see you active again :)

Fat-Zer, 2023-03-06:
I've dug a bit dipper into the issue, and it looks like the message is originating from KDE, namely the [kxmlgui][1] lib... This library when being loaded injects its own event filter which produces an error if a conflicting shortcut is triggered. This prevents FreeCAD from handling the conflict by it's own means... According to the backtrace (see bellow) it looks like that the loading chain of the kde library is `[Any python-heavy workbench] -> Gui::UiLoader::UiLoader() -> QUiLoader::QUiLoader() -> loads all available plugins in can find in libraryPaths()`

The problem is that QUiLoader has no direct means of preventing external plugins from being loaded. The only solution I can think of is to reset `QApplication::libraryPaths()` for the time of instancing of a `UiLoader`...

The issue with inability to change shortcuts is probably a separate one.

I probably should report both bugs separately from this issue...

-----

```
#0  _k_installConflictDetector() () at /usr/src/debug/kde-frameworks/kxmlgui-5.102.0/kxmlgui-5.102.0/src/kactionconflictdetector.cpp:53
#1  0x00007ffff1648047 in qAddPreRoutine(void (*)()) () at /usr/lib64/libQt5Core.so.5
#2  0x00007fff3d8159f0 in _k_installConflictDetector_ctor_function() () at /usr/src/debug/kde-frameworks/kxmlgui-5.102.0/kxmlgui-5.102.0/src/kactionconflictdetector.cpp:57
#3  0x00007fff3d815a03 in (anonymous namespace)::_k_installConflictDetector_ctor_function_ctor_class_::_k_installConflictDetector_ctor_function_ctor_class_()
    (this=0x7fff3d90c338 <(anonymous namespace)::_k_installConflictDetector_ctor_function_ctor_instance_>) at /usr/src/debug/kde-frameworks/kxmlgui-5.102.0/kxmlgui-5.102.0/src/kactionconflictdetector.cpp:57
#4  0x00007fff3d815b18 in __static_initialization_and_destruction_0(int, int) (__initialize_p=1, __priority=65535) at /usr/src/debug/kde-frameworks/kxmlgui-5.102.0/kxmlgui-5.102.0/src/kactionconflictdetector.cpp:57
#5  0x00007fff3d815b2e in _GLOBAL__sub_I_kactionconflictdetector.cpp(void) () at /usr/src/debug/kde-frameworks/kxmlgui-5.102.0/kxmlgui-5.102.0_build/src/KF5XmlGui_autogen/include/kactionconflictdetector.moc:93
#6  0x00007ffff7fceafe in call_init () at /lib64/ld-linux-x86-64.so.2
#7  0x00007ffff7fcebe4 in _dl_init () at /lib64/ld-linux-x86-64.so.2
#8  0x00007ffff0e29d2d in _dl_catch_exception () at /lib64/libc.so.6
#9  0x00007ffff7fd4e86 in dl_open_worker () at /lib64/ld-linux-x86-64.so.2
#10 0x00007ffff0e29cd0 in _dl_catch_exception () at /lib64/libc.so.6
#11 0x00007ffff7fd520b in _dl_open () at /lib64/ld-linux-x86-64.so.2
#12 0x00007ffff0d60f48 in dlopen_doit () at /lib64/libc.so.6
#13 0x00007ffff0e29cd0 in _dl_catch_exception () at /lib64/libc.so.6
#14 0x00007ffff0e29d8f in _dl_catch_error () at /lib64/libc.so.6
#15 0x00007ffff0d60a26 in _dlerror_run () at /lib64/libc.so.6
#16 0x00007ffff0d61001 in dlopen () at /lib64/libc.so.6
#17 0x00007ffff1640c11 in  () at /usr/lib64/libQt5Core.so.5
#18 0x00007ffff163ba75 in  () at /usr/lib64/libQt5Core.so.5
#19 0x00007ffff163c083 in  () at /usr/lib64/libQt5Core.so.5
#20 0x00007ffff1633516 in QPluginLoader::load() () at /usr/lib64/libQt5Core.so.5
#21 0x00007ffff6ce6460 in QFormInternal::QFormBuilder::updateCustomWidgets() (this=0x555559bb03d0) at /var/tmp/portage/dev-qt/designer-5.15.8/work/qttools-everywhere-src-5.15.8/src/designer/src/lib/uilib/formbuilder.cpp:491
#22 0x00007ffff6ce5987 in QFormInternal::QFormBuilder::setPluginPath(QStringList const&) (this=this@entry=0x555559bb03d0, pluginPaths=...)
    at /var/tmp/portage/dev-qt/designer-5.15.8/work/qttools-everywhere-src-5.15.8/src/designer/src/lib/uilib/formbuilder.cpp:452
#23 0x00007ffff6cc94c3 in QUiLoader::QUiLoader(QObject*) (this=<optimized out>, parent=<optimized out>) at /var/tmp/portage/dev-qt/designer-5.15.8/work/qttools-everywhere-src-5.15.8/src/designer/src/uitools/quiloader.cpp:677
#24 0x00007ffff6c2b6a5 in Gui::UiLoader::UiLoader(QObject*) (this=0x555556f35838, parent=0x0) at /home/alexander/projects/free-cad_code/src/Gui/UiLoader.cpp:489
#25 0x00007ffff6c2bc64 in Gui::UiLoaderPy::UiLoaderPy() (this=0x555556f35820) at /home/alexander/projects/free-cad_code/src/Gui/UiLoader.cpp:546
#26 0x00007ffff6c2b885 in Gui::UiLoaderPy::PyMake(_typeobject*, _object*, _object*) (args=()) at /home/alexander/projects/free-cad_code/src/Gui/UiLoader.cpp:516
....
```

------

> Hi @Fat-Zer nice to see you active again :)

hi, thanks... haven't respond earlier to not flood bug comments with flame... at least without bug update))...

 [1]: https://github.com/KDE/kxmlgui/blob/master/src/kactionconflictdetector.cpp

luzpaz, 2023-03-06:
Cool, nice sleuthing! please feel free to link upstream bug reports to this ticket so we can track the progress

Fat-Zer, 2023-03-09:
> 3. GUI doesn't provide an ability to modify a shortcut

As for this claim, it's a bit embracing, but I overlooked how it's supposed to work: You should enter a new shortcut into the field helpfully called `New shortcut` and only then press `Apply`...
So my bad, the claim was wrong... At most I would suggest to add one or two tooltips to avoid such confusion in case I'm not such a unique dumbass...

Fat-Zer, 2023-03-10:
I filled in upstream bugs:

- Qt: https://bugreports.qt.io/browse/QTBUG-111856
- KDE: https://bugs.kde.org/show_bug.cgi?id=467130

Though I believe we still should implement the workaround, because those are relatively minor and a bit controversial issues (more design flaws than bugs)... So I don't expect they will get a lot of attention... And even after that the fixes won't propagate into a release anytime soon...

closed 2023-03-18 by wwmayer