FreeCAD/FreeCAD #8710: [Feature Request] Add the outside of the Sketch on PD (Relates with issue #8706)

leoheck, 2023-03-01:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://github.com/FreeCAD/FreeCAD/issues/8706

### Subproject(s) affected?

PartDesign

### Idea description

This is a similar idea to the https://github.com/FreeCAD/FreseCAD/issues/8706
However, the idea is to pad (or add mass) from outside of a Sketch.

The existing design would be used as a bounding box between the Sketch and the infinity.
In this way, it would not work if used in as the First feature of the Body.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-01:
It's a stronger case if you provide an use case for this feature on what difficulty it would be solving, according to the new contribution process an issue (including feature requests) should be described in terms of a problem that has to be solved, in that spirit please describe what situation is difficult when this feature is not present?

Do note please that I am not advocating against this feature but merely asking for good justification to improve the request.

leoheck, 2023-03-01:
Sure. I don't have any use case for this. I just mentioned it  since it is a complementarity case of the other issue. I created this to remove tis discussions from the other issue so we can close it due to the fact I don't know a use case for this.

luzpaz, 2023-03-02:
Re-opening. Usecase was given by @FlachyJoe in:  
https://github.com/FreeCAD/FreeCAD/issues/8706#issuecomment-1449918242  
https://github.com/FreeCAD/FreeCAD/issues/8706#issuecomment-1450044863

adrianinsaval, 2023-03-02:
That's not really an usecase

leoheck, 2023-03-02:
Exactly, that is an example on how it would work. Use case is to show why it is important to waist someone's lifetime to implement this.

leoheck, 2023-03-02:
The author of the second item/link said this

[image]

luzpaz, 2023-03-02:
I linked to @FlachyJoe directly so they could comment. But point taken, closing FR

closed 2023-03-02 by luzpaz

FlachyJoe, 2023-03-02:
> I linked to @FlachyJoe directly so they could comment. But point taken, closing FR

I agree. I can't find an useful extrusion of outer sketch.