FreeCAD/FreeCAD #8711: [Issue] Polygon changing shape when distance constraint changed

mathiaslec, 2023-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32146 (Git) Snap 636
Build type: Release
Branch: master
Hash: 04d494cb75e180a4805feedc6d9c96e2ec306f97
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Issue description

Hi all,
In Sketcher, when I change the vertical or horizontal distance of a polygon from another point, it changes the shape of the polygon or raises an error.

How to reproduce the problem : 

1. In sketcher, create a sketch (for example XY-plane);
2. Draw a polygon (for example an hexagon), and add all constraints: horizontal constraint for 1 of the polygon sides, 5 mm in diameter for the polygon, vertical constraints for the polygon to the origin, and 10 mm vertical distance from the center of the polygon to the origin;
3. Change the last constraint (vertical distance from the origin) to another value : 10 mm to 30 mm.

Note that it can have different behaviors : 

- for some values the polygon shape is not modified;
- for some other values the shape is modified (from hexagon to square for example);
- and for some others it raises errors. For example :

```
14:48:27  Updating geometry: Error build geometry(0): Both points are equal
14:48:27  Invalid solution from DogLeg solver.
```

I could reproduce this on Windows 11 and FreeCAD 0.19.

Thanks for the help.
[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-02:
@AjinkyaDahale do you mind reproducing?

utkarshraj779, 2023-03-02:
i think that 1st draw a circle after that making plygon inside the circle which should be benificial for making and observing the distance

AjinkyaDahale, 2023-03-03:
Can confirm. This seems to be yet another degenerate case: the constraint is "all endpoints should be on the circle, and all lines should be equal". One way for this to happen is for all edges to be zero length.

maxwxyz, 2024-03-30:
Also can reproduce on 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.4
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.9

```

mathiaslec, 2025-01-06:
The issue is still relevant.

hyarion, 2026-03-28:
@PaddleStroke could you try to clean up the your comment? It's quite difficult to read in this form. 🙏

PaddleStroke, 2026-03-28:
done

luzpaz, 2026-09-03:
Bunped PR for visibility