FreeCAD/FreeCAD #8717: [Problem]

pespin, 2023-03-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
freecad version: "0.20.2 Revision 29603 (Git)"
package installed from Archlinux: community/freecad 0.20.2-4
Using: Archlinux (Linux), with Enlightenment as Display manager under X11
HW: Lenovo Thinkpad x280 (intel GPU)

$ ldd /usr/lib/freecad/bin/FreeCAD
	linux-vdso.so.1 (0x00007ffca5ed5000)
	libFreeCADGui.so => /usr/lib/freecad/lib/libFreeCADGui.so (0x00007efd0ec00000)
	libFreeCADApp.so => /usr/lib/freecad/lib/libFreeCADApp.so (0x00007efd0e600000)
	libFreeCADBase.so => /usr/lib/freecad/lib/libFreeCADBase.so (0x00007efd0e403000)
	libpython3.10.so.1.0 => /usr/lib/libpython3.10.so.1.0 (0x00007efd0e000000)
	libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5 (0x00007efd0d800000)
	libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5 (0x00007efd0d000000)
	libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x00007efd0ca00000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007efd0c600000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007efd0fdf3000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007efd0c419000)
	libCoin.so.80 => /usr/lib/libCoin.so.80 (0x00007efd0ba00000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007efd0fd6d000)
	libQt5PrintSupport.so.5 => /usr/lib/libQt5PrintSupport.so.5 (0x00007efd0eb90000)
	libQt5Svg.so.5 => /usr/lib/libQt5Svg.so.5 (0x00007efd0eb37000)
	libspnav.so.0 => /usr/lib/libspnav.so.0 (0x00007efd0fd62000)
	libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 => /usr/lib/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007efd0eb06000)
	libboost_filesystem.so.1.81.0 => /usr/lib/libboost_filesystem.so.1.81.0 (0x00007efd0fd3f000)
	libxerces-c-3.2.so => /usr/lib/libxerces-c-3.2.so (0x00007efd0b600000)
	libQt5Xml.so.5 => /usr/lib/libQt5Xml.so.5 (0x00007efd0e3c1000)
	libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 => /usr/lib/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007efd0dfaf000)
	libQt5Network.so.5 => /usr/lib/libQt5Network.so.5 (0x00007efd0d68f000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007efd0dec7000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007efd0fe58000)
	libboost_program_options.so.1.81.0 => /usr/lib/libboost_program_options.so.1.81.0 (0x00007efd0cfb9000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007efd0eaec000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007efd0cf80000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007efd0c906000)
	libmd4c.so.0 => /usr/lib/libmd4c.so.0 (0x00007efd0ead6000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007efd0c337000)
	libdouble-conversion.so.3 => /usr/lib/libdouble-conversion.so.3 (0x00007efd0d679000)
	libicui18n.so.72 => /usr/lib/libicui18n.so.72 (0x00007efd0b200000)
	libicuuc.so.72 => /usr/lib/libicuuc.so.72 (0x00007efd0ae00000)
	libpcre2-16.so.0 => /usr/lib/libpcre2-16.so.0 (0x00007efd0c878000)
	libzstd.so.1 => /usr/lib/libzstd.so.1 (0x00007efd0b130000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007efd0acc1000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007efd0ab7e000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007efd0cf55000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007efd0cf23000)
	libOpenGL.so.0 => /usr/lib/libOpenGL.so.0 (0x00007efd0c84d000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007efd0b548000)
	libQt5Qml.so.5 => /usr/lib/libQt5Qml.so.5 (0x00007efd0a600000)
	libboost_atomic.so.1.81.0 => /usr/lib/libboost_atomic.so.1.81.0 (0x00007efd0e3b5000)
	libnsl.so.3 => /usr/lib/libnsl.so.3 (0x00007efd0debe000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0x00007efd0b085000)
	libicudata.so.72 => /usr/lib/libicudata.so.72 (0x00007efd08800000)
	libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2 (0x00007efd0b9ac000)
	libssl.so.3 => /usr/lib/libssl.so.3 (0x00007efd0aade000)
	libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x00007efd08200000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007efd08723000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007efd0c314000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007efd0d66d000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007efd080b8000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007efd0b979000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007efd0b063000)
	libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0 (0x00007efd0aa43000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007efd0b038000)
	libtirpc.so.3 => /usr/lib/libtirpc.so.3 (0x00007efd0b00b000)
	libnghttp2.so.14 => /usr/lib/libnghttp2.so.14 (0x00007efd0aa17000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007efd0a5de000)
	libssh2.so.1 => /usr/lib/libssh2.so.1 (0x00007efd086e1000)
	libpsl.so.5 => /usr/lib/libpsl.so.5 (0x00007efd0c839000)
	libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1 (0x00007efd0c306000)
	libkrb5.so.3 => /usr/lib/libkrb5.so.3 (0x00007efd07fe0000)
	libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x00007efd086b3000)
	libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x00007efd0b973000)
	libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x00007efd0b965000)
	libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x00007efd0b541000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007efd0aa05000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007efd086a0000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007efd0867a000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007efd0b53c000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007efd0b532000)
	libunistring.so.5 => /usr/lib/libunistring.so.5 (0x00007efd07e26000)
	libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1 (0x00007efd08657000)
```

### Subproject(s) affected?

None

### Problem description

I installed today and started freecad for first time, and it's totally unusable due to the Qt menus being displayed/rendered in a wrong way. There seems to be some sort of scaling/clipping issue.
I attach a couple screenshots (one before installing qt5-styleplugins package and one afterwards) showing the display errors.

I talked with another Archlinux user and he's seeing the same behavior.

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-02:
It looks like you might be able to get at the Help menu there -- can you try it and see if you can paste in the text that's copied when you click the Help->About FreeCAD->Copy to clipboard button? I'd like to see exactly what Qt this is compiled against, and if you have any addons that might be interfering.

pespin, 2023-03-02:
When I click "Help" from the upper menu bar (it seems the place where I have to click in the bar to get to open its dropdown menu is more left than the text, hence showing some sort of scaling issue), I press the "Help" option in the drop down menu, I get this popup window (again showing some sort of error with scaling):
[image]
Since it cannot be resized I'm not sure what the test says or the button says.
Pressing the left button seems to close the window.
Pressing the Right button takes me to this other window (which can be resized):
[image]

btw I see nowhere that "About FreeCAD"  you mention. Is it at the end of the dropdown menu? maybe due to the scaling issues I cannot see it.

xtemp09, 2023-03-02:
Hooray! I reproduced this bug! FreeCAD is installed from official Arch repositories.

[image]
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/10874919/FreeCAD.log)

chennes, 2023-03-02:
Do you happen to know the exact version of Qt that is being used for Arch?

pespin, 2023-03-02:
```
$ yay -Ss qt5 | grep Installed
aur/qt5-styleplugins 5.0.0.20170311-35 (+70 0.89) (Out-of-date: 2023-02-06) (Installed)
extra/qt5-xmlpatterns 5.15.8-1 (1.1 MiB 5.4 MiB) [qt qt5] (Installed)
extra/qt5-x11extras 5.15.8-1 (14.8 KiB 40.8 KiB) [qt qt5] (Installed)
extra/qt5-webengine 5.15.12-3 (48.7 MiB 151.5 MiB) [qt qt5] (Installed)
extra/qt5-webchannel 5.15.8+kde+r3-1 (75.0 KiB 232.5 KiB) [qt qt5] (Installed)
extra/qt5-wayland 5.15.8+kde+r59-1 (1.1 MiB 6.1 MiB) [qt qt5] (Installed)
extra/qt5-translations 5.15.8-1 (1.9 MiB 14.7 MiB) [qt qt5] (Installed)
extra/qt5-tools 5.15.8+kde+r1-3 (5.3 MiB 15.5 MiB) [qt qt5] (Installed)
extra/qt5-svg 5.15.8+kde+r8-1 (149.1 KiB 499.0 KiB) [qt qt5] (Installed)
extra/qt5-multimedia 5.15.8+kde+r2-1 (763.0 KiB 3.4 MiB) [qt qt5] (Installed)
extra/qt5-location 5.15.8+kde+r3-1 (1.2 MiB 5.4 MiB) [qt qt5] (Installed)
extra/qt5-imageformats 5.15.8+kde+r9-1 (54.2 KiB 209.5 KiB) [qt qt5] (Installed)
extra/qt5-declarative 5.15.8+kde+r24-2 (4.8 MiB 23.5 MiB) [qt qt5] (Installed)
extra/qt5-base 5.15.8+kde+r181-1 (13.0 MiB 63.0 MiB) [qt qt5] (Installed)
extra/qscintilla-qt5 2.13.4-1 (1.1 MiB 6.5 MiB) (Installed)
extra/qca-qt5 2.3.5-2 (759.6 KiB 2.9 MiB) (Installed)
extra/python-pyqt5-sip 12.11.1-1 (73.9 KiB 212.3 KiB) (Installed)
extra/python-pyqt5 5.15.9-1 (4.3 MiB 28.2 MiB) [pyqt5] (Installed)
```
I think it also shows up in FreeCAD.log which @xtemp09 uploaded earlier.

pespin, 2023-03-02:
I just talked on IRC #archlinu chatroom with 2 people who tried to reproduce the issue (both using up to date Archlinux like me, so same package versions, Qt 5.15.8 and freecad 0.20.2).
One tested with both xfce4 and  moksha (enligthnment 17 fork), using amd driver
Another tested with xfce4, nouveau, xorg

None of them could reproduce the bug above.
I'm personally using an intel GPU:
`00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07) (prog-if 00 [VGA controller])`

pespin, 2023-03-02:
It's definitely some sort of Qt scaling issue. Using `QT_SCALE_FACTOR=1.3 freecad` then it works fine
Actually for anything >=1.3 it works fine, < 1.3 I see the buggy behavior I described in before.

carstenhaitzler, 2023-03-02:
E Will set xft.dpi and xsettings dpi based on what it wants apps to scale to. E sets it to the same. Apps should then uniformly scale using that. i suspect freecad is picking and choosing different values for scaling from different places and rendering some things with one scale and some with other scales (eg scissor clips with one, actual content geometry with another). or using a gl viewport transform but not also then adjusting the scissors... but either way it amounts to the same - not consistently using the same value given. e doesn't set the qt scale factor - preferring the above instead.

xtemp09, 2023-03-03:
I entered `export QT_SCALE_FACTOR=1.3` and then started FreeCAD. Everything seems to be alright.

carstenhaitzler, 2023-03-03:
so .. it's a bug in freecad with math for smaller scale factors. probably scaling some things to the dpi given but then forcing a minimum size on things in other ways thus being "off" with some elements of the rendering pipeline. might be a qt bug too - not sure. not an area i'm really familiar with (other than having written a bit of qt code long ago and knowing how to build toolkits and how to render :))

chennes, 2023-03-03:
It's fascinating that the dividing line is 1.3 -- I might have guessed a rounding error or something if it was 1.5, but 1.3 seems very arbitrary.

pespin, 2023-03-03:
@chennes I tried 1.2, 1.25 and 1.29 and those are presenting the bug. 1.3 works fine, I have been using that for several hours without problems.

pespin, 2023-03-03:
Actually, 1.27 definetly shows the error. With 1.28 and 1.29 I'm not that sure since the difference may be minimal.

wwmayer, 2023-03-08:
In the Debian ML this problem has been reported too: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=986821
On the mentioned systems the issue has appeared after changing the DPI value from the default of 96 to e.g. 80.
By doing so I was able to reproduce it on my system (XFCE) as well.

Since other Qt apps like QtCreator or Designer don't suffer from this problem I started to look at the code and found the line:
`QGuiApplication::setHighDpiScaleFactorRoundingPolicy(Qt::HighDpiScaleFactorRoundingPolicy::PassThrough);`
Commenting it out solved the problem for me.

https://github.com/FreeCAD/FreeCAD/pull/8783

adrianinsaval, 2023-03-08:
this issue has been repeatedly reported on the forum, almost always the user had a dpi setting smaller than 96, most often observed among xfce or mate users

pespin, 2023-03-09:
I confirm the issue is fixed running current freecad master (0adc473d53732c3a1e1e5f6c6dc1468d2a65cff5) built from archlinux AUR.

So I guess the issue was fixed by https://github.com/FreeCAD/FreeCAD/pull/8783

closed 2023-03-09 by pespin