FreeCAD/FreeCAD #8719: [Issue] Rotated sketch when drawing on face

Glitchy-Tozier, 2023-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=73602

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32164 (Git) Snap 638
Build type: Release
Branch: master
Hash: cc574749fa8af962670f77f26548bc71046361f0
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Issue description

1. Open project
2. Select the Face2 on the top (see screenshot)
3. Create a sketch
4. Although the sketch can be attached with the mode "Flat Face" the sketch appears to be orthogonal to the selected face.
5. It should be noted that the Face2 is a BsplineSurface.

### Anything else?

1. A more descriptive title is probably needed.
2. GitHub doesn't like the `.FCStd` file extension, so here's a link to the project file posted in the forum: https://forum.freecad.org/download/file.php?id=217053&sid=a816270ff637c8358db62afd9b82b9c3
3. Screenshot: [image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-12:
It seems that this issue should be reopened: https://github.com/FreeCAD/FreeCAD/pull/14372#issuecomment-2162365434

FEA-eng, 2024-08-01:
@maxwxyz Can you reopen this issue as explained above?