FreeCAD/FreeCAD #8721: [Issue]

PLChris, 2023-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=664650#p664644

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.54
  * sheetmetal 0.2.61
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Moving a B-spline in Sketchre leaves the control points in their old place. To reproduce:
- open the (unzipped) file
[moveSpline.zip](https://github.com/FreeCAD/FreeCAD/files/10871294/moveSpline.zip)
- select the B-spline
- invoke the Sketcher Move tool
- move it to the left

-> this is the result where the control points are still in their old positions:
[image]
The solver message is "Solver failed to converge".

Moving this B-spline worked in 0.20.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-03-02:
This also happens for other curves with internal geometries, including (arcs of) conics.

AjinkyaDahale, 2023-03-02:
@abdullahtahiriyo what is our current policy on the move tool? Can you confirm it's still to be supported or do you plan to deprecate it for something else?

luzpaz, 2023-03-02:
Hey @PLChris do you mind giving this ticket a name? TIA

abdullahtahiriyo, 2023-03-02:
I think all geometries move properly if the internal alignment geometry is selected in addition. 

The tool predates complex geometry, so it was never designed to support it.

I do not have a current policy for the tool. We should ask the community about it. If it is used and is to remain, then we should fix the issues (Ctrl+Z, ...). If it is not used and is to be removed, then we should do so.

AjinkyaDahale, 2023-03-03:
> We should ask the community about it. If it is used and is to remain, then we should fix the issues (Ctrl+Z, ...). If it is not used and is to be removed, then we should do so.

@chennes like WIP Wednesdays we could have some "CTA XDays" where we ask the community to come and give opinions about some decisions that need to be made.

chennes, 2023-03-03:
I don't know if the's an every-week thing there, but I could certainly do something like a blog post with a quick poll in it. We have lots of different communities out there (Facebook groups, Mastodon, Twitter, LinkedIn, the Forums, the blog, etc.) -- for really big things we could coordinate a poll between all of them.

PaddleStroke, 2023-03-16:
I think move copy and clone tools will be deprecated. In the tool setting dev branch I made a 'translate' tool that was working well to replace move + current rectangular array.
Besides there is also the 'group dragging' feature that is somewhere in the pipeline. So move will be pretty useless when you'll be able to drag.

PaddleStroke, 2023-03-16:
@AjinkyaDahale not about this specifically since the tool settings is still quite far off in the future.

maxwxyz, 2024-02-21:
@PaddleStroke is this still relevant after your implementation?

PaddleStroke, 2024-02-21:
I have not touched bspline tool yet. So i can't say for sure.

maxwxyz, 2024-03-30:
Closing this as with the new array transform tool also the control points are moved.

closed 2024-03-30 by maxwxyz