FreeCAD/FreeCAD #8723: [Problem]

kberkut, 2023-03-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

The dimensions disappear when you zoom in on the sketch
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-02:
Please add a legit name for ths ticket

luzpaz, 2023-03-03:
@kberkut would you mind trying to reproduce this on the dev version? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

kberkut, 2023-03-20:
> @kberkut would you mind trying to reproduce this on the dev version? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Not work.
For first time i'm see very big size dimension marker. For second redact i'm see very small and it not zoom. The more I increase the smaller the dimensional letters become.
[image]
[image]

[code]
OS: Arch Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Help 1.0.3
  * sheetmetal 0.2.61
  * LCInterlocking
[/code]

luzpaz, 2023-03-20:
Can you also provide an example file? (please zip it and drag in to the github textbox to attach)

kberkut, 2023-03-20:
This mapping happens every other time, not always.
[Project_name.zip](https://github.com/FreeCAD/FreeCAD/files/11018009/Project_name.zip)

maxwxyz, 2024-03-30:
I cannot reproduce, also with the provided file.
Is this still relevant @kberkut? Please try this in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

closed 2024-06-15 by maxwxyz