FreeCAD/FreeCAD #8733: [Problem] format library not working in files in Base, OK in App

berniev, 2023-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Current
```

### Subproject(s) affected?

None

### Problem description

Tried in App::Annotation.cpp and test: worked fine.

Tried in Base::Unit.cpp and got error building test:

`[4/6] Linking CXX shared library lib/libFreeCADBase.dylib
FAILED: lib/libFreeCADBase.dylib 
: && /usr/local/opt/llvm/bin/clang++ -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -g -DFC_DEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -mmacosx-version-min=12.6 -dynamiclib -Wl,-headerpad_max_install_names -Wl,-undefined,error -o lib/libFreeCADBase.dylib -install_name @rpath/libFreeCADBase.dylib src/Base/CMakeFiles/FreeCADBase.dir/FreeCADBase_autogen/mocs_compilation.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxxextensions.c.o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxx_extensions.cxx.o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxxsupport.cxx.o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/IndirectPythonInterface.cxx.o src/Base/CMakeFiles/FreeCADBase.dir/__/CXX/cxx_exceptions.cxx.o src/Base/CMakeFiles/FreeCADBase.dir/Axis.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/AxisPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Base64.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/BaseClass.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/BaseClassPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/BindingManager.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/BoundBoxPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Builder3D.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Console.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ConsoleObserver.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/CoordinateSystem.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/CoordinateSystemPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Debugger.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/DualQuaternion.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Exception.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ExceptionFactory.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Factory.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/FileInfo.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/FileTemplate.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/FutureWatcherProgress.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/GeometryPyCXX.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Handle.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/InputSource.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Interpreter.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Matrix.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/MatrixPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/MemDebug.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Mutex.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Parameter.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ParameterPy.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Persistence.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PersistencePyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Placement.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PlacementPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PrecisionPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ProgressIndicatorPy.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PyExport.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PyObjectBase.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/QtTools.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Reader.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Rotation.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/RotationPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Sequencer.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/SmartPtrPy.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Stream.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Swap.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/swigpyrun.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/TimeInfo.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Tools.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Tools2D.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Tools3D.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Translate.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Type.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/TypePyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Uuid.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Vector3D.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/VectorPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ViewProj.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Writer.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/XMLTools.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/ZipHeader.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsApi.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsApiPy.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchema.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaInternal.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaMKS.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaImperial1.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaCentimeters.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaMmMin.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitsSchemaFemMilliMeterNewton.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Quantity.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/QuantityPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/Unit.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/UnitPyImp.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/PyTools.c.o src/Base/CMakeFiles/FreeCADBase.dir/PreCompiled.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/collcoll.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/deflateoutputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/fcoll.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/fcollexceptions.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/fileentry.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/filepath.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/filterinputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/filteroutputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/gzipoutputstream.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/gzipoutputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/inflateinputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipfile.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/ziphead.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstreambuf.cpp.o src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstream.cpp.o -F/usr/local/lib  -L/usr/local/Cellar/open-mpi/4.1.4_2/lib   -L/usr/local/opt/libevent/lib -Wl,-rpath,/usr/local/Cellar/open-mpi/4.1.4_2/lib -Wl,-rpath,/usr/local/opt/libevent/lib -Wl,-rpath,/usr/local/lib  /usr/local/lib/libhdf5.dylib  /usr/local/lib/libsz.dylib  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libz.tbd  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libdl.tbd  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libm.tbd  -lmpi  /usr/local/opt/xerces-c/lib/libxerces-c.dylib  /usr/local/lib/libboost_filesystem-mt.dylib  /usr/local/lib/libboost_program_options-mt.dylib  /usr/local/lib/libboost_regex-mt.dylib  /usr/local/lib/libboost_system-mt.dylib  /usr/local/lib/libboost_thread-mt.dylib  /usr/local/lib/libboost_date_time-mt.dylib  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libz.tbd  /usr/local/lib/QtCore.framework/Versions/A/QtCore  /usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/lib/libpython3.10.dylib  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libdl.tbd  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libm.tbd  -lmpi  /usr/local/opt/xerces-c/lib/libxerces-c.dylib  /usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/lib/libpython3.10.dylib  /usr/local/lib/libboost_atomic-mt.dylib  -framework IOKit  -framework DiskArbitration && :
Undefined symbols for architecture x86_64:
  "fmt::v9::vformat(fmt::v9::basic_string_view<char>, fmt::v9::basic_format_args<fmt::v9::basic_format_context<fmt::v9::appender, char> >)", referenced from:
      Base::Unit::getString() const in Unit.cpp.o
ld: symbol(s) not found for architecture x86_64
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-04:
Did you add `fmt::fmt` to FreeCADBase_LIBS?

berniev, 2023-03-04:
No. Just reporting a problem.

chennes, 2023-03-04:
Is this an existing test, or a new one you are writing? I believe all existing tests are compiling on my systems.

berniev, 2023-03-04:
I don't know what has changed, but seems to be ok now. Infuriating because I went to a lot of trouble to test. Nothing consciously changed here.

closed 2023-03-04 by berniev