FreeCAD/FreeCAD #8734: [Bug] not correct displaying of recent macros names

FreeCAD-Tools, 2023-03-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
```

### Subproject(s) affected?

None

### Issue description

File "Test.FC.Macro.FCMacro" shows as simply "Test"

Everything that comes after the dot in the name is not displayed.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-04:
@Syres916 want to take a stab at this ?

Syres916, 2023-03-06:
Looks like a really simple change of line 1168 in src\Gui\Action.cpp from:

`        recentFiles[index]->setText(QString::fromLatin1("%1 %2").arg(index+1).arg(fi.baseName()));`

to:

`        recentFiles[index]->setText(QString::fromLatin1("%1 %2").arg(index+1).arg(fi.completeBaseName()));`

If the OP wants to create the PR then fine with me.

luzpaz, 2023-03-06:
@FreeCAD-Tools want to make a PR ?

FreeCAD-Tools, 2023-03-06:
@luzpaz @Syres916

PR should be done by the one who solves the problem. In any case, I won't be able to debug this C++ code for checking. But I don't mind if you add me as a co-author to the PR. Just add text below to end of PR message as that do in [https://github.com/shaise/FreeCAD_FastenersWB/pull/249](https://github.com/shaise/FreeCAD_FastenersWB/pull/249) . This should work.

closed 2023-03-09 by luzpaz

donovaly, 2023-03-09:
The fix will be in FreeCAD 0.20.3