FreeCAD/FreeCAD #8741: Mesh Design "Unwrap Mesh" broken in 0.21 ?

PhiJu5, 2023-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * AirPlaneDesign 0.4.0
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Curves.backup1678010057.868822 0.6.8 (Disabled)
  * fasteners 0.4.55
  * MeshRemodel 1.8919.0
  * Plot 2022.4.17
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Use attached Unwrap.FCStd (in zip file), mesh "Solid" than "Unwrap Mesh" the mesh fails.

Traceback (most recent call last):
  File "E:\FreeCad21\Mod\MeshPart\MeshFlatteningCommand.py", line 58, in Activated
    boundaries = flattener.getFlatBoundaryNodes()

Unable to convert function return value to a Python type! The signature was
	(self: flatmesh.FaceUnwrapper) -> std::vector<Eigen::Matrix<double,-1,3,0,-1,3>,std::allocator<Eigen::Matrix<double,-1,3,0,-1,3> > >

Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>,
<pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic
conversions are optional and require extra headers to be included
when compiling your pybind11 module.

Doing same with 020 works.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * AirPlaneDesign 0.4.0
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Curves.backup1678010057.868822 0.6.8 (Disabled)
  * fasteners 0.4.55
  * MeshRemodel 1.8919.0
  * Plot 2022.4.17
[/code]

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-05:
> Did you forget to `#include <pybind11/stl.h>`? Or `<pybind11/complex.h>`,
`<pybind11/functional.h>`, `<pybind11/chrono.h>`, etc. Some automatic
conversions are optional and require extra headers to be included
when compiling your pybind11 module.

@donovaly, perhaps these libs were removed in one of the lib 'purges' ?

donovaly, 2023-03-09:
> @donovaly, perhaps these libs were removed in one of the lib 'purges' ?

No. One must only set in CMake these options
* BUILD_FLAT_MESH
* FREECAD_USE_PYBIND11

After these have been set to True, a complete recompilation might be necessary to get the feature.

donovaly, 2023-03-09:
@PhiJu5 do you compile by yourself? If not, from where do you get your build?

PhiJu5, 2023-03-09:
No, download from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
FreeCAD-0.21.0dev-32110-Installer.exe

donovaly, 2023-03-09:
> No, download from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds FreeCAD-0.21.0dev-32110-Installer.exe

Thanks.
I indeed made a mistake once I cleaned the includes. #8799 will fix this.
I will provide a new weekly installer tomorrow with the fix included.

closed 2023-03-09 by donovaly

donovaly, 2023-03-11:
> I will provide a new weekly installer tomorrow with the fix included.

Here it is: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-32258-Installer.exe

PR #8799 (2023-03-09): [MeshPart] fix unwrap feature
- fixes #8741
- in commit d62d41a53b66af1e824e the pybind headers were accidentally removed
commit bedfd8f7 (2023-03-09): [MeshPart] fix unwrap feature

- fixes #8741
- in commit d62d41a53b66af1e824e the pybind headers were accidentally removed