FreeCAD/FreeCAD #8743: [Issue] [Path] List index out of range

GrzegorzKrug, 2023-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Issue description

Pocket3D fails on whole shape. This issue exist for both models normal and upsidedown.
I though this issue is only related to `3Dpocket` and `Adaptive`, but it looks like its also for `Profile` so it may be common issue.
Traceback:
```
15:50:38  <Path.Area> Area.cpp(1619): skipped 1 sub shapes during projection
15:50:38  <Path.Area> Area.cpp(1459): empty projection
15:50:38  Traceback (most recent call last):
  File "P:\Program Files\FreeCad\Mod\Path\PathScripts\PathUtils.py", line 71, in new_function
    res = function(*args, **kwargs)
  File "P:\Program Files\FreeCad\Mod\Path\PathScripts\PathOp.py", line 820, in execute
    result = self.opExecute(obj)
  File "P:\Program Files\FreeCad\Mod\Path\PathScripts\PathAreaOp.py", line 366, in opExecute
    aOS = self.areaOpShapes(obj)
  File "P:\Program Files\FreeCad\Mod\Path\PathScripts\PathPocket.py", line 248, in areaOpShapes
    env = PathUtils.getEnvelope(
  File "P:\Program Files\FreeCad\Mod\Path\PathScripts\PathUtils.py", line 265, in getEnvelope
    sec = area.makeSections(heights=[0.0], project=True)[0].getShape()
<class 'IndexError'>: list index out of range
15:50:38  Recompute failed! Please check report view.

```
[image]

### Anything else?

Attaching save file.
[Miska-4Liscie.zip (Poket error)](https://github.com/FreeCAD/FreeCAD/files/10892081/Miska-4Liscie.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-07:
@mlampert do you mind weighing-in ?

luzpaz, 2023-04-17:
@sliptonic do you have a moment to check this out ?

JohnOCFII, 2025-01-16:
This remains an issue. Please don't close it.

luzpaz, 2025-02-01:
@maxwxyz are you keeping an eye on the closed tickets ?

maxwxyz, 2025-02-01:
Not all of them

luzpaz, 2025-02-01:
There will be a need to circle back and see if any legit tickets were closed, right ?

GrzegorzKrug, 2025-02-03:
At this point, you should maybe introduce some tag to issues, that will prevent from auto closing :P 
unless you want to track over 2k issues

maxwxyz, 2025-02-03:
There is a no-auto close issues and feature requests are not going to be closed. Closed issues are not all being revisited.

If somebody experiences the same issue, one can comment in the closed issue and it can be reopened or a new issue can be created.

We have so many open issues and if no one touched an issue for several years, either it only affects very few users or no one cares to fix it. If the issue is bad enough, the issue itself will have more attention as more people are commenting on it, therefore the action will also not close it.

The problem is, that we have not enough people who can fix stuff and no product owner like person per workbench prioritizing bugs and feature requests and assigning them to developers working on them, as currently there are only volunteers contributing. For that the tracker with open issues is too big so issues which are not that relevant will get lost quickly.

We have discussed this action twice in the developers meetings and the conclusion was, we simply let it close issues for these reasons. I'm sure you an find the minutes on the developer meeting repo.

sliptonic, 2025-05-20:
I tried retesting this after the upgrade of libarea/clipper today.  I don't see the index out of range error.  
@luzpaz can you confirm and close?

JohnOCFII, 2025-05-20:
Re-opening that particular file still lists the index out of range message, but since that file was created in 0.20, I don't know if that is still a valid test case.

[image]

I tried a fresh file, and did not receive the List Index out of range (although the pathing seemed odd - not sure if that is related to the recent change in clipper).  This was a self-built version of FreeCAD using Pixi build process that I built this morning.  I'll put that into a different / new issue if it happens repeatably.

[image]

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41802 (Git)
Build date: 2025/05/20 13:59:52
Build type: Unknown
Branch: main
Hash: 448976dfc5931b066d7683d8ff453641f1f53bfb
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * AddonManager 2025.5.1a
  * Telemetry 1.0.1
```

sliptonic, 2025-07-06:
I think we close this issue.  
The original file predates jobfixer.  Even with it run, I can't duplicate the problem.  
OP, if there is still something to address, please comment and we will reopen.

closed 2025-07-06 by sliptonic