FreeCAD/FreeCAD #8744: Chamfer is not working anymor

yoyo325, 2023-03-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Curves 0.6.5
  * fasteners 0.4.54
  * ThreadProfile 1.84.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

When using the Chamfer knob in Parts Design as well as in Parts the programm is not responding anymore.

Already deleted and reinstalled Freecad but nothing happens. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-05:
Not working anymore? What is the exact issue you're observing ?

yoyo325, 2023-03-05:
Sorry , problem solved. Wrong use of datum plane which caused the error
Thanks anyhow for response

closed 2023-03-05 by yoyo325