FreeCAD/FreeCAD #8749: [Issue] FreeCAD crashes when using Draft_Edit

azuk, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=73045

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: recent master, see hash
Build type: Debug
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * Render 2023.2.2
  * BIM 2021.12.0
  * Reinforcement
  * Manipulator 1.5.0
  * Assembly4.backup1662941442.2770898
  * parts_library
  * Assembly4 0.12.6
  * fasteners 0.4.55
  * Help 1.0.3
  * Silk 1.0.0
  * dodo 1.0.0
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

Draft

### Issue description

In addition to the forum link given above, there is another forum thread about the same issue here: https://forum.freecad.org/viewtopic.php?t=76516&sid=2b48d079e2c3bb0f01f4c7c05023b41c

Steps to reproduce:
1. Create a new document.
2. Switch to Draft workbench and select Top view
3. Add a horizontal draft line
4. Double click the created object in the tree view
5. Select the leftmost point of the line with mouse
6. Restrict change to just x coordinate with shift and move cursor to task dialog and select the "Local delta X" input
7. Enter a positive value that moves the endpoint a little bit to the right and hit return.
8. Observe that FreeCAD crashes. Stacktrace below.

The problem cannot be reproduced by moving the line endpoint with mouse, you need to enter the delta X value to the dialog.

On a fast computer the problem may never happen. Even on a slower computer it may not happen every time. One can artificially slow down key event processing in debugger (by adding a breakpoint that does nothing but logs something and then continues, for example) to reproduce it more easily.

Stacktrace from one attempt: 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x151bd36fb090]
#1  0x151bd14da376 in QQmlData::isSignalConnected(QAbstractDeclarativeData*, QObject const*, int) from /lib/x86_64-linux-gnu/libQt5Qml.so.5+0x6
#2  0x151bd3d58a6e in QMetaObject::activate(QObject*, int, int, void**) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x6e
#3  0x151bd489ac8d in QWidgetLineControl::processKeyEvent(QKeyEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x2cd
#4  0x151bd488d99f in QLineEdit::keyPressEvent(QKeyEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1f
#5  0x151bd7f5dfe2 in Gui::InputField::keyPressEvent(QKeyEvent*) from /usr/local/freecad/lib/libFreeCADGui.so+0x382
#6  0x151bd478a77d in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x74d
#7  0x151bd4893d32 in QLineEdit::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xb2
#8  0x151bd4747a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#9  0x151bd475181e in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa5e
#10  0x151bd794914a in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/freecad/lib/libFreeCADGui.so+0x112
#11  0x151bd3d2d80a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#12  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc216) [0x151bd47a9216]
#13  0x151bd4747a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#14  0x151bd47510f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#15  0x151bd794914a in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/freecad/lib/libFreeCADGui.so+0x112
#16  0x151bd3d2d80a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#17  0x151bd4114b2b in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x16b
#18  0x151bd411a0f1 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x241
#19  0x151bd40f435b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#20  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x151bcd73f32e]
#21  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x151bd19d917d]
#22  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x151bd19d9400]
#23  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x151bd19d94a3]
#24  0x151bd3d85435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#25  0x151bd3d2c3ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#26  0x151bd3d34116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#27  0x151bd77f5956 in Gui::Application::runApplication() from /usr/local/freecad/lib/libFreeCADGui.so+0x28f2
#28  /usr/local/freecad/bin/FreeCAD(+0x12987) [0x55fccbdbe987]
#29  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x151bd36dc083]
#30  /usr/local/freecad/bin/FreeCAD(+0x11c2e) [0x55fccbdbdc2e]
```

I bisected the problem to commit fc14567d3424e19e8a3a7d333e253f46214ca469.

Edit.endEditing in Mod/Draft/draftguitools/gui_edit.py calls DraftToolBar.editUi and before that commit editUi called self.taskUi which did this (partial copy&paste):

```
            self.isTaskOn = True
            todo.delay(FreeCADGui.Control.closeDialog,None)
            self.baseWidget = DraftBaseWidget()
            self.layout = QtGui.QVBoxLayout(self.baseWidget)
            self.setupToolBar(task=True)
            self.retranslateUi(self.baseWidget)
            self.panel = DraftTaskPanel(self.baseWidget,extra)
            todo.delay(FreeCADGui.Control.showDialog,self.panel)
```

Now editUi calls self.makeDumbTask which does this (class TaskPanel definition not included):

```
        FreeCADGui.Control.closeDialog()
        panel = TaskPanel(extra, on_close_call)
        FreeCADGui.Control.showDialog(panel)
```

So, observe that both the old code and the new close the current dialog and show a new one. However, the old code does both of those things using the todo.delay mechanism.

Right after filing this issue report, I will try to submit a PR that wraps closeDialog and showDialog calls in makeDumbTask in todo.delay, effectively restoring the behaviour before fc14567d3424e19e8a3a7d333e253f46214ca469. That mitigates the problem so that I can no longer reproduce it when running FreeCAD normally on my slower computer.

However, I can still reproduce the problem when running FreeCAD under a debugger (with breakpoint-logging in a key event processing method, just to introduce extra delay). Seems to be timing sensitive and while todo.delay helps, it's not a proper fix. So I'd recommend leaving this issue open despite the mitigation PR.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-10:
Roy assigned to you. Feel free to unassign if not appropriate.

Roy-043, 2023-03-11:
Fixed with #8750.

closed 2023-03-11 by Roy-043