FreeCAD/FreeCAD #8751: [Problem] Open a file or a project crash frecad

herveparis752011, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Curves 0.5.12
  * fasteners 0.4.21
  * sheetmetal 0.2.49
  * ThreadProfile 1.84.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

If I try to open a file, or import, a window open blank, and freecad shutdown.
It wasn't append with my previous version of freecad (guess 0.19)
What could I do ?
Thank for your help, and excuse my poor english

Hervé

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

herveparis752011, 2023-03-06:
BTW I am under windows 10

luzpaz, 2023-03-06:
Hello. We can't help you unless you provide more info. Because of the language barrier, you may want to get help in the [FreeCAD French forum](https://forum.freecad.org/viewforum.php?style=4&f=12).

herveparis752011, 2023-03-06:
Thank you for changing the title.
I think I'm be able to understand an answer in english.
What's more infos do you want ?
I start Freecad 0.20.2, create a new empty document, and a new body. Then If I try to open an existing freecad's  file i have a suddenly shutdown. I still asked this problem in the french forum, but I don't have a satisfying answer.

I hope this can help you

adrianinsaval, 2023-03-28:
what is an "alias" folder?
did you try the development version?

windows is shit because it doesn't show the output of apps started from terminal, please from cmd launch freecad with the --write-log option and reproduce the issue, then upload the generated log file that you should find at %AppData%/FreeCAD

herveparis752011, 2023-03-29:
« Alias » mean a shortcut folder. (.lnk)

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

Envoyé le :mardi 28 mars 2023 22:19
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>

luzpaz, 2023-03-29:
@herveparis752011 please respond on github instead of via email (since it's generating a lot of spam) :point_up:

herveparis752011, 2023-03-29:
Sorry
HC

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

Envoyé le :mercredi 29 mars 2023 16:14

luzpaz, 2023-03-29:
Lol...you responded via email to my request not to respond via email 🤦

luzpaz, 2023-04-09:
So the problem here is the .lnk file?

herveparis752011, 2023-04-09:
Yes apparently. 
I had also completely uninstalled the soft, and re install it.
But I still have the problem, so I had created a new folder on my desk, and it's work fine.
Hope this can help

xtemp09, 2023-04-09:
herveparis752011, how can one reproduce the problem?

herveparis752011, 2023-04-09:
create a folder in a usb drive, then create a shortcut of this folder on your desktop.
By the way I haven't try to do the same thing with my HDD.
Hope this can help

xtemp09, 2023-04-10:
Not reproducible on

```ruby
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32662 (Git)
Build type: Release
Branch: master
Hash: 312fe9b5c822126b251ca1c3cfb55dde0de1920c
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

luzpaz, 2023-06-10:
@herveparis752011 can you test this on development version 0.21.x  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and report back ?

adrianinsaval, 2023-06-11:
I can't reproduce this at all so I can only assume a problem on OP's system, maybe a damaged usb? or maybe something related to locale? We would need exact reproduction steps, example file and the path where the file is stored, how and where the shortcut is created and how to trigger the crash on FreeCAD.

adrianinsaval, 2023-08-22:
No response, not reproducible, not actionable. Closing

closed 2023-08-22 by adrianinsaval