FreeCAD/FreeCAD #8754: [Feature Request] PD: Adding guide curves to loft

69Piero, 2023-03-06:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76467

### Subproject(s) affected?

PartDesign

### Idea description

Currently the loft between two profiles is straight and out from user's control, so the feature is weak. You need ability to guide the loft by paths (sketches) so the user can manage "the way in which a profile turns into another one."

Sweep could be a solution, but it supports just one guide curve so doesn't solve the issue

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

SV001R, 2023-12-11:
I need this feature too. Making guided loft solid or surface is weakpoint of FC's loft feature.

AjinkyaDahale, 2023-12-21:
So some opinions here:
1. Since we already have a sweep (also called pipe sometimes in PD source code), it is that method which should have these additional "guides", unless a loft can do something sweep cannot (this maybe possible given that we use different methods for loft and sweep, viz. `BRepOffsetAPI_ThruSections` and `BRepOffsetAPI_MakePipeShell`). There may also be a case for combining Loft and Sweep, but let's not go there just yet.
2. Sweep already has a single guide. There is also an "auxiliary spine", though it is used to [define the _normal_ at every cross section](https://dev.opencascade.org/doc/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html#a91843c65b1f6cecd9b7296c43a8b7c96). So, to summarize, if we view a sweep as a union of interpolated faces, the "spine" dictates where the "relevant" point at each interpolated cross section (this term might not be the best descriptor) lies, and the "auxiliary spine" dictates how that interpolated cross section should be oriented. The only means we have to control the shape of the interpolated cross section is to add more in-between shapes.
3. It is theoretically possible to imagine multiple spines dictating the shape of the cross sections. To take it to the extreme, this can also theoretically be done continuously along the whole wire (requiring uncountably infinite spines). but this is not supported in OCCT itself, at least in `BRepOffsetAPI_MakePipeShell`. There may be other methods, which are used by the curve workbenches that give better control.

Right now I see two options:
1. Find out those methods from the curve WB's and use them here. OR
2. Suggest that the user move to these workbenches if the item they want to build is that complicated.