FreeCAD/FreeCAD #8755: [Issue] declaration different

maddin200, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20 source code
```

### Subproject(s) affected?

None

### Issue description

decalaration different 
src/Mod/Path/libarea/kurve/Construction.cpp line 699

code:
Definition:
double	atn360(double dx, double dy);	// dx dy vs dy dx
declaration:
// distance
	double atn360(double dy, double dx) {
		// angle 0 to 2pi
		double ang = atan2(dy, dx);
		return ((ang < 0)? 2 * PI + ang : ang);
	}

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-06:
Please elaborate by improving the title of this ticket and what you are trying to point out.

adrianinsaval, 2023-03-28:
when looking at source code please do so on the master branch not a release branch and please use proper links to the affected source code:
https://github.com/FreeCAD/FreeCAD/blob/87b0893044f16da41aa84a500d7faaa895a953aa/src/Mod/Path/libarea/kurve/geometry.h#L604
https://github.com/FreeCAD/FreeCAD/blob/87b0893044f16da41aa84a500d7faaa895a953aa/src/Mod/Path/libarea/kurve/Construction.cpp#L699-L703

@sliptonic is this a legitimate issue?

maddin200, 2023-07-03:
Nevertheless the names should be either corrected or omitted.

luzpaz, 2024-08-27:
Added reference link directly into source in the original ticket.

closed 2025-01-19 by github-actions[bot]