FreeCAD/FreeCAD #8758: [Crash] In PartDesign when doing Toothbrush Head Stand tutorial

dperetti, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Dracula 0.0.2
[/code]

(also tried a native version installed via brew)
```

### Subproject(s) affected?

None

### Problem description

Giving FreeCad a try, starting with this [tutorial](https://wiki.freecad.org/Toothbrush_Head_Stand).
Crashed every single time (about 5 or 6) when accomplishing this step:
[image]

Tried the arm version installed with brew and the x86 version.
Same issue at about the same time.
Did it one more time while typing this report. Crashed again (see attached log).

### Anything else?

[freecad.log](https://github.com/FreeCAD/FreeCAD/files/10900207/freecad.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-08:
@dperetti can you attach a file (zip the .FCStd file and drag it in to the ticket) so testers can quickly reproduce ?

dperetti, 2023-03-08:
I switched to the dev branch and could not make it happen again. Is it worth keeping the investigations? If it is I'll switch back to the stable release and do what you asked.

luzpaz, 2023-03-08:
@donovaly thoughts?

donovaly, 2023-03-09:
If the crash can be reproduced in an official build for FreeCAD 0.20.2, the bug should stay open, otherwise we can close it.

adrianinsaval, 2023-03-28:
I think we don't have official arm builds for 0.20 and I would consider emulation unsupported. I think this one should be marked as solved in 0.21 so that we remember to close it upon release.

It would be a good idea to get some feedback from other mac arm users on this though, maybe @chennes or @oursland ?

oursland, 2023-03-28:
I am only aware of a single "official" macOS ARM build and that's the [Weekly Build from January 7th, 2023](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds).  The builds installed from `brew` are all Rosetta 2-emulated x86_64 builds, and I can confirm crashes are not uncommon in this configuration.

macOS Apple Silicon builds are within reach, which along with Python 3.11 will definitely improve performance on macOS.  I cannot say if this will make things more stable, however I will try to reproduce the error on my current ARM-native builds.

oursland, 2023-03-29:
I am unable to reproduce the crash on my development macOS ARM64 build.

[image]

I suspect the issue is either an emulation issue or an older bug that has been resolved by recent development.

adrianinsaval, 2023-03-29:
Thank you for testing. I think it's safe to close this one

luzpaz, 2023-03-29:
Thx for the follow-up!
Closing

closed 2023-03-29 by luzpaz