FreeCAD/FreeCAD #8760: [Issue] Invalid mesh generated

burunduk3, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
_built from sources at commit 1a0801a23f917f534d4dabe8c956df4cd206e2a2_
[code]
OS: Gentoo Linux (/usr/share/xsessions/xmonad-wtf)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32107 (Git)
Build type: Unknown
Branch: master
Hash: 1032fe4468b5c41e579b133ffcb0f702e24c6982
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.0.3, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Subproject: Mesh design

When trying to create a mesh from part, the result appears to be not a correct mesh (open edges, wrong surfaces, see the `bad-mesh.stl` in the archive I've attached).

I've tried to make a small example which reproduces the issue, see `repro-bad-mesh.FCStd` in the archive.

[repro-bad-mesh.zip](https://github.com/FreeCAD/FreeCAD/files/10903387/repro-bad-mesh.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-30:
Should we close ticket ?

burunduk3, 2023-05-30:
Thanks, yes, I will try looking at the geometry some time.

I still have a question of how did I get to this point (I think I just mirrored an element twice and did a cut), but when I'll be able to name specific actions which lead to the invalid geometry, I can always create a new issue for that.

luzpaz, 2023-05-30:
> I still have a question of how did I get to this point (I think I just mirrored an element twice and did a cut), but when I'll be able to name specific actions which lead to the invalid geometry, I can always create a new issue for that.  

Indeed. If you can recreate the workflow that leads to the invalid geometry then please feel to open a new ticket (and reference this one in it so there is a continuity). TIA!

closed 2023-05-30 by luzpaz