FreeCAD/FreeCAD #8762: [Spreadsheet in Part Design] <CLEAR and OK-BUTTON>

Manuelzwei, 2023-03-06:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

-

### Subproject(s) affected?

PartDesign

### Idea description

Hello
FreeCAD 0.20.1, Win10 64bit

[image]

When I enter an alias from a spreadsheet, the cursor jumped to the CHANCEL-button and not to the OK-button as default.
So I often clear the value unintentionally.

Could you please set it up so that the OK-button is activated when the value or alias in the input field is valid?

Thanks a lot!

### Anything else?

I think this affects other input fields as well

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-07:
Can you test this on v0.21.x ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Manuelzwei, 2023-03-09:
luzpaz, thank you for your Answer!

No, the same effect as in 0.20.

[image]

My cursor jump to the ["LÖSCHEN"] eg. "CANCEL"-field...

[image]

...but you can't see my cursor on the screenshot.

When I press the "ENTER"-button on the keyboard, "OK" was selected, it works. 
But I have my hand on the mouse ans I don't want to switch to the keyboard.

In Win xx there was an option by "setting for the mouse"

"Einstellungen" -->> "Weitere Mausoptionen" -->> "Zeigeroptionen" -->> "Zur Standardschaltfläche springen":

[image]

Manuelzwei, 2023-03-09:
Can you change the place from these buttons?
"OK" first than the "CANCEL"-button?

I select the alias with my mouse between the first and the fifth letters from the alias, so my mouse is on the position near - or on top of - the "CANCEL"-button.

If I select the alias '10cm' away, then the mouse position is somewhere in the space and then I have to move to the button and click.

luzpaz, 2023-03-10:
Looks like related UI code is in the general vicinity of:  
https://github.com/FreeCAD/FreeCAD/blob/f99e76056e9bd98f86db368f8cc726d8eead0a1b/src/Gui/DlgExpressionInput.ui#L26

I tried to move the OK button over in QtDesigner and it changes a lot more of the .ui file than what seems necessary.

adrianinsaval, 2023-03-10:
At most we can move both icons to the right... swapping is not desirable as it's different from standard button location

adrianinsaval, 2023-03-10:
This is also a very niche case IMO, you write parameters with the keyboard, why don't you just select the desired object with the keyboard and press enter without changing focus from the field.

luzpaz, 2023-03-10:
Just our of curiosity, are other dialogs also following this Cancel | OK scheme ?

Manuelzwei, 2023-03-10:
> This is also a very niche case IMO, you write parameters with the keyboard, why don't you just select the desired object with the keyboard and press enter without changing focus from the field.

I type the values in the spreadsheet and select then all with the mouse. 
I rarely use the keyboard.

Manuelzwei, 2023-03-10:
> At most we can move both icons to the right... swapping is not desirable as it's different from standard button location

Yes, that is a way

Manuelzwei, 2023-03-10:
> Just out of curiosity, are other dialogs also following this Cancel | OK scheme ? (also does the 1st click of the tab highlight the 'Cancel' button first ?)

I will try to find somewhere.

THANK you ALL together!

luzpaz, 2023-03-15:
[image]

@yorikvanhavre do you have a moment to do this ?