FreeCAD/FreeCAD #8764: Addon Manager uses wrong URL for Macro Icons

jvelie, 2023-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76605

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32198 (Git)
Build type: Release
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.8.16, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
```

### Subproject(s) affected?

Addon Manager

### Issue description

User reports the following error

> running addon manager i find an update for the macro CenterofMass but its installation fails. that's the report:
>
> 07:33:25 Impossibile installare la macro CenterOfMass
> 07:33:25 Failed to create C:\Users\PF93F~1.GRA\AppData\Local\Temp\tmpwvss0hdp\https://wiki.freecadweb.org/images/4/41

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-08:
Related to #8734 ?

chennes, 2023-03-08:
Nope.

closed 2023-03-24 by chennes

donovaly, 2023-03-24:
The fix will be available in FreeCAD 0.20.3.

PR #9018 (2023-03-24): Addon Manager: Fix wiki macro icon cleanup
Fixes #8764.

- [X]  ThisPull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR