FreeCAD/FreeCAD #8765: [Issue] TopoNaming Phase 1 (card 1 of 4) - create IndexedName class

JohnAD, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming algorithm would benefit from a type to easily store and manipulate the short reference names of the elements.

## description

Create an IndexedName.cpp class that is used to save the name of an element. It will later by used by the `MappedElement` class which will be used by the `ElementMap` class.

One benefit of this change is to "simplify" the renaming/numbering of faces/etc. For example, it provides a `+=` operator to increment the number part.

It stores the "short name" of the element, such as "Face1".

For context, see issue https://github.com/FreeCAD/FreeCAD/issues/8432

## notes

Instead of a simple string, it is a string/integer combination. The string is the "type" of element. The integer is an number on it's suffix. So, instead of storing "face3", this class stores "face" and 3.

The `toString()` function resolves the class back into a simple string.

suggested private data members in the class:

```cpp
    const char* ;
    int;
```

most of the work has been done already in the branch at `src/App/MappedElement.h`

mostly, just extract this into a separate file and add good unit tests to ensure that these core functions are very reliable.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-08:
I've started putting together a first PR for this issue, and I have a question about the `toString` function:
```
const char * toString(std::string & s) const
    {
        // Note! s is not cleared on purpose.
        std::size_t offset = s.size();
        s += this->type;
        if (this->index > 0) {
            s += std::to_string(this->index);
        }
        return s.c_str() + offset;
    }
```
The function name seems a little obscure to me here. It's not returning a `std::string`, and not only is it returning a `const char *` but it's also modifying a parameter. I'm betting there is a better name for this function (and the parameter `s`). @JohnAD any suggestion?

chennes, 2023-03-08:
@realthunder can you explain the choice of using C-style code here: https://github.com/FreeCAD/FreeCAD/blob/355a3275b320d5b3c2f8a54df0d7c429899ba58e/src/App/IndexedName.cpp#L105
```
    for (const char* type : types) {
        int j = 0;
        for (const char* n = name, *t = type; *n; ++n) {
            if (*n != *t || j >= i)
                break;
            ++j;
            ++t;
            if (!*t) {
                this->type = type;
                return;
            }
        }
    }
```
This is basically a startswith algorithm, right? (I know we can't use the std::startswith yet, it wasn't added until C++20, but there are more idiomatic ways of doing this).

JohnAD, 2023-03-09:
@chennes ,

In nearly every case I saw `const char * toString(std::string & s) const` being used, either the returned value was being used and the parameter's changed value ignored, such as:

```cpp
std::string s;
Py::String name(res.index.toString(s));
```

or the opposite, the changed string was used to build a longer string, but the returned value was discarded:

```cpp
for(auto &v : ret) {
            tmp.clear();
            v.index.toString(tmp);
// ...
```

I recommend we create to two functions instead.

First, a more traditional `toString()` with no parameters.

Second, and something better named that only modifies the passed ref such as `void appendName(std::string &s)`.

Technically, if a caller needed both, the combined op saves memory since the returned pointer points to the interior of the passed string. But, IMO, if we encountered a scenario where that is important (I don't see one in the code right now), we can created a 3rd function with a wild-and-long-name like `getNameAndAlsoAppendIt`

Just my opinion though.

JohnAD, 2023-03-09:
On a philosophical level, I'm generally uncomfortable with functions that **both* modify passed values and return a new value. I'd rather they do one or the other. But, that is just my philosophy, sometimes narrow scenarios happen where it makes sense to do both. I just don't see this as one of those scenarios.

chennes, 2023-03-09:
As a general rule in modern code I don't use `const char *` for _anything_ -- the miniscule performance gain never strikes me as being worth the risks. I don't have a great sense here for whether it really makes sense to store `type` as a `const char *`. @JohnAD do you have a feeling about whether we could just use a `std::string` instead?

JohnAD, 2023-03-09:
I agree, let's use `std::string` instead. We can always change course later if we really have to, but let's try the cleaner path first.

realthunder, 2023-03-09:
> This is basically a startswith algorithm, right?

@chennes Yes, but with additional requirement of the first `i` number of characters in `name` to match the entire string of `type`.

> The function name seems a little obscure to me here. It's not returning a `std::string`, and not only is it returning a `const char *` but it's also modifying a parameter.

The idiom for this kind of function is to convert to a string but for the caller to provide buffer. The returned the value is just for convenience so that the caller can use right away. I am sure you've seen this type of idiom before. The string may contain other content the caller may want to preserve.

Regarding the performance thing, please note that the current version of topo naming encoding scheme is the result after several iterations. I did use plain std::string in the initial version. It was okay for moderate modeling, but got worse pretty fast when reaching certain size, especially when there are compound shapes. The problem is the extra memory it requires. This is the reason I use `QByteArray`, split out common prefix, and have hierarchical names for delayed name expansion, etc.

realthunder, 2023-03-09:
> I agree, let's use `std::string` instead. We can always change course later if we really have to, but let's try the cleaner path first.

There is a reason of using `const char*` here. It makes `IndexedName` an extremely light weight structure, which is important, as there can be literally millions of instances of this kind of highly repetitive data. Also note that `IndexedName` does not simply reuse the input string pointer, `IndexedName::set()` function will try to lookup the type from an explicitly supplied type array that is supplied by the caller, which is assumed to be persistent, and hence safe for reuse. If the type cannot be found, and the caller allows this 'unexpected' type, `IndexedName` will store the name in an internal `set` to avoid duplication.

chennes, 2023-03-09:
> Regarding the performance thing, please note that the current version of topo naming encoding scheme is the result after several iterations. I did use plain std::string in the initial version. It [...]

Makes sense -- @JohnAD I'm inclined in this case to stick with char * despite the extra complications, since @realthunder has done the tests here.

adrianinsaval, 2023-03-09:
I suggest adding a comment to indicate this is done for critical performance reason so that a future dev doesn't waste time trying to make this "cleaner"

chennes, 2023-03-09:
The `std::string` will have four extra bytes, every time, even for short strings (I think). Regarding the buffer-supplied version: I still think it would be more idiomatic to have two functions, one with a clear user-provided buffer, and one without. Nothing about that call, function name, or function signature says to me "this appends a string to your user-provided buffer".

chennes, 2023-03-09:
Something like this?
```
void appendToStringBuffer(std::string & buffer) const
    {
        buffer += this->type;
        if (this->index > 0)
            buffer += std::to_string(this->index);
    }

    std::string toString() const
    {
        std::string result;
        this->appendToStringBuffer(result);
        return result;
    }
```

JohnAD, 2023-03-09:
Looks good!

chennes, 2023-03-09:
@realthunder is the intent here (called when the set insertion of the type name succeeds):
```
void mutate() const
    {
        QByteArray copy;
        copy.append(bytes.constData(), bytes.size());
        bytes = copy;
    }
```
to ensure that the QByteArray in the unordered_set has a unique copy of the data? Or is this doing something else?

chennes, 2023-03-10:
Another question (either for @realthunder or anyone else who is familiar with how IndexedName is used in practice): is the following test an accurate representation of the expected behavior? (Given an `allowedTypes` vector that contains `EDGE`):
```
    auto indexedName = Data::IndexedName("EDGE19", allowedTypes, false);
    EXPECT_STREQ(indexedName.getType(), "EDGE");
    indexedName = Data::IndexedName("EDGES_ARE_REALLY_GREAT19", allowedTypes, false);
    EXPECT_STREQ(indexedName.getType(), "EDGE");
```

realthunder, 2023-03-10:
> is the intent in the following code to ensure that the QByteArray in the unordered_set has a unique copy of the data? Or is this doing something else?

@chennes `ByteArray::mutate()` is called right after inserting into `IndexedName::NameSet`. To avoid wasted buffer allocation, I use QByteArray::fromRawData() to wrap the caller supplied const string pointer to attempt an insertion. If it fails, no buffer deallocation is needed. If it succeeds, `mutate()` will be called to ensure its member QByteArray owns the string.

> is the following test an accurate representation of the expected behavior

The second test case shall fail. Because the last argument `allowOthers` is false, and since `EDGES_ARE_REALLY_GREAT` is  not in `allowedTypes`, `indexedName.getType()` will return an empty string, and `indexedName.isNull()` will return true.

chennes, 2023-03-10:
> The second test case shall fail. Because the last argument `allowOthers` is false, and since `EDGES_ARE_REALLY_GREAT` is not in `allowedTypes`, `indexedName.getType()` will return an empty string, and `indexedName.isNull()` will return true.

This does not appear to be what actually happens. I've extracted your matching algorithm to its own function for testing:
```
bool IndexedName::originalNameMatchAlgorithm(int i, std::vector<const char *> types, const char *name)
{
    for (const char * type : types) {
        int j=0;
        for (const char *n=name, *t=type; *n; ++n) {
            if (*n != *t || j >= i)
                break;
            ++j;
            ++t;
            if (!*t) {
                return true;
            }
        }
    }
    return false;
}
```
(it's identical to your original code, except it returns true or false instead of directly setting the name variable). Given the following test:
```
TEST_F(IndexedNameTest, originalNameMatchAlgorithm)
{
    const char* testName = "EDGE19";
    bool nameMatched = Data::IndexedName::originalNameMatchAlgorithm(4, allowedTypes, testName);
    EXPECT_TRUE(nameMatched);
    testName = "EDGES_ARE_REALLY_GREAT19";
    nameMatched = Data::IndexedName::originalNameMatchAlgorithm(22, allowedTypes, testName);
    EXPECT_FALSE(nameMatched);
}
```
This test fails, because `nameMatched` is `true` in both cases. As far as I can tell the algorithm you wrote is a "starts_with" algorithm, not a "strings are equal" algorithm. So if the requested name **starts with** any of the strings in the vector, the value from the vector is used, and the remainder of the name is discarded. If the intention is to actually ensure that the _entire_ name (minus its index) matches, we should just be using `std::strcmp`, rather than doing a manual letter-by-letter comparison.

PR #8776 (2023-03-15): App: Toponaming indexed name
Fixes #8765 -- extracts IndexedName into its own header and source files, performs clang-format and clang-tidy cleanups on it, refactors to clarify and correct some issues, and adds unit tests.

- [X]  This Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
closed 2023-03-15 by chennes