FreeCAD/FreeCAD #8766: [Issue] TopoNaming Phase 1 (card 2 of 4) - create MappedName class

JohnAD, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming algorithm would benefit from a type to easily store and manipulate the name serialized from the element mapping.

## description

Create MappedName.cs to store and manipulate the more-complex mapped-name used by elements (and the toponaming algorithm).

For context, see issue https://github.com/FreeCAD/FreeCAD/issues/8432

## notes

private data members are defined as follow in the branch:

```cpp
    QByteArray data; 
    QByteArray postfix; 
    bool raw;
```

where `data` could contain something like "Face2" which is either the parent (?? or the actual name if not postfix ??)
and `postfix` could contain something like ";:M2;FUS;:Hb4d:8,F" which is a "description" how we got here.

The methods of determining the contents of those fields is done in phase 2 in the ElementMap class; this class should be simpler.

I personally recommend using a more descriptive name than `data`, but those are private elements so the naming is not critical.

Most of the work already done in `src/App/MappedElement.*`

mostly, just extract this into a separate file and add unit tests to ensure that these core functions are very reliable.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-12:
@realthunder the `MappedName` class includes some functionality to make a copy of a "slice" of the mapped name, potentially overlapping from the data to the postfix, e.g.
```
    auto temp = Data::MappedName::fromRawData(QByteArray("TESTTEST", 8));
    temp.append("ABCDEFGHIJKLM"); //postfix
    auto mappedName = Data::MappedName::fromRawData(temp, 2, 13);
```
This seems like undesirable behavior to me: can you describe a circumstance where we want to take just part of the data, and just part of the postfix? I didn't see any obvious uses in the code itself, but if you've got a link I'd be interested to see what's going on there.

chennes, 2023-03-16:
Regarding the `MappedName::compact` function: first, it seems to me that `compress` would be a better name for the function, since in English "compact" is commonly used as both a verb and a noun. "Compress" is rarely used as a noun, so to me seems clearer.

Second: the function is declared as `const`, but then:
```
void MappedName::compact() const
{
    auto self = const_cast<MappedName*>(this);

    // Does stuff with "self" that changes this
}
```

This seems highly suspect to me: what is the reason for declaring this a `const` function? If it absolutely _has_ to be const, why not declare the internal data as `mutable`?

chennes, 2023-03-17:
In this function:
```
const char* toPrefixedString(std::string& buf) const
    {
        if (!toIndexedName()) {
            buf += ComplexGeoData::elementMapPrefix();
        }
        appendToBuffer(buf);
        return buf.c_str();
    }
```
why is `toIndexedName()` the condition that governs whether or not to prefix the string?