FreeCAD/FreeCAD #8767: [Issue] TopoNaming Phase 1 (card 3 of 4) - create MappedElement class

JohnAD, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming algorithm would benefit from a type to easily store and manipulate element naming as a whole.

## description

Create teh MappedElement.cpp class that is used to save the name parts of an element. Steps 1 and 2 must be complete first.

For context, see issue https://github.com/FreeCAD/FreeCAD/issues/8432

## notes

It contains the current IndexedName and the generated MappedName.

most of the work has been done already in the branch at `src/App/MappedElement.h`.

The comparator `ElementNameComp` and the classes `HistoryItem` and `MappedChildElements` will be added elsewhere in phase 2 when ElementMap code is built out.

mostly, just extract this into a separate file and add good unit tests to ensure that these core functions are very reliable.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-15:
I'm going to start in on this one next.

chennes, 2023-03-15:
Is it important to have both of these constructors? Can we just pick one as the standard?
```
    MappedElement(const IndexedName & idx, const MappedName & n);
    MappedElement(const MappedName & n, const IndexedName & idx);
```

chennes, 2023-03-15:
I am going to stub it out and then use gMock for the actual unit tests.

AjinkyaDahale, 2023-03-16:
> I am going to stub MappedName out and then use gMock for the actual unit tests.

@chennes please let me know once this is ready. I'm trying to simply copy the relevant files and for now the dependencies keep coming.

chennes, 2023-03-16:
Yeah, I changed my mind about this: it would have required making the whole class virtual, and I don't think that's a good idea. I've just copied in the work that @Pesc0 has done so far, and won't add it to git.

PR #8974 (2023-03-25): App/Toponaming: Add MappedElement
Closes #8767.

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
closed 2023-03-25 by chennes