FreeCAD/FreeCAD #8768: [Issue]

JohnAD, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming is more easily worked on if the descriptive elements of it are gathered in one place for easy reference.

## description

The goal here is to create a common place for the consts and other names that will be used by toponaming as a whole.

`ElementMap.h`

At first, this will be mainly the strings used for serialization of the MappedName's elements. As common point of reference are discovered related to the toponaming ElementMap algorithm, this will be a good place to place them.

Later, in phase 2, this header file will be used with the associated ElementMap.cpp class.

For context, see issue https://github.com/FreeCAD/FreeCAD/issues/8432

## notes

Go ahead and move the contents of src/Mod/Part/App/TopoShapeOpCode.h to here. Include the conditional directives.

For the following items, see ComplexGeoData.cpp, TopoShapeEx.cpp.

| postfix (elementMap key) | meaning       |
|----------------|---------------|
| `:H`           | tag (local)   |
| `:C`           | child element |
| `:X`           | external tag  |
| `:I`           | index         |
| `D`            | duplicate     |
| `:T`           | deprecated tag (uses decimal) |
| `:M`           | mod           |
| `:MG`            | modgen        |
| `:G`           | gen           |
| `:U`           | upper         |
| `:L`           | lower         |

(above edited based on RealThunder's feedback)

For example,

```cpp
const std::string &ComplexGeoData::tagPostfix() {
    static std::string postfix(elementMapPrefix() + ":H");
    return postfix;
}
```

in ComplexGeoData.cpp chould be moved here to simply be something like:

```cpp
const std::string POSTFIX_FOR_TAG = ";:H";
```

I recommend we also go ahead and move `elementMapPrefix()` to a const as it simply returns a semicolon (always).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-03-15:
I'll give this a try. Is this a standard practice in C++ to create a static bool

```
    ElementMap()
    {
        static bool inited;
        if (!inited) {
        ...
        }
    }
```

There are a bunch of `connect`s in the `...` and maybe it can be placed somewhere else?

AjinkyaDahale, 2023-03-16:
@JohnAD are you sure we should be moving elements from `Part` to `App`? Usually `Part` is already a dependency wherever these are used and I don't really see TopoShape being used a lot in `App` side.

PR #8923 (2023-03-23): [Core] Create `ElementMap.h` and tests
`ElementMap.h` for now contains a list of constants which will be used in the `ElementMap` class in a future phase of toponaming solution.

Intended to fix #8768.