FreeCAD/FreeCAD #8774: [Arch] Remove the "Untitled" IFC type

yorikvanhavre, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=665093#p665093

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32205 (Git)
Build type: Unknown
Branch: master
Hash: 8381ea079ddadf7c582dcfafeb8a6f9c4c004108
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BIM 2021.12.0
  * FreeCAD-NativeIFC 2022.10.20
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Arch

### Issue description

The IFCType property of Arch object allows for a known IFC type, or "Undefined". The problem is, that Undefined type gets currently mapped to different things on export, depending on the object.

The "Undefined" entry should be removed, to "force" each object to define a valid default type.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-10:
@yorikvanhavre will this make it in to v0.21 release ?

yorikvanhavre, 2023-06-12:
Yes! On it!

PR #9765 (2023-06-19): Arch remove undefined type
Removes the "Undefined" IFC type and uses "IfcBuildingElementProxy" as default instead, which is the de-facto "undefined" type of the IFC standard.

Fixes #8774 

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://gith [... cut]
commit 599b2499 (2023-06-19): Arch: Removed Undefined IFC type - fixes #8774
closed 2023-06-19 by yorikvanhavre