FreeCAD/FreeCAD #8777: [Issue] PartDesign Hole: wrong depth for countersink

PLChris, 2023-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76650

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.54
  * sheetmetal 0.2.61
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

None

### Issue description

For non default countersink angles the depth is not correct.
[image]

The hole in the center uses only defaults, except
- HoleCutType=Countersink
- HoleCutDepth=2mm

The hole at the right uses additionally 
- HoleCutCountersinkAngle=45°

The image shows that the depths are different, that for the second hole is wrong.
[countersinkDepth.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/10918767/countersinkDepth.FCStd.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-08:
the code that generates the shape does not take the depth parameter into account
https://github.com/FreeCAD/FreeCAD/blob/61f1310aa59c95a7c866ac619b8699932d94da46/src/Mod/PartDesign/App/FeatureHole.cpp#L1748-L1750

I can't find any logic that relates the countersink depth and diameter in the code, any hints @donovaly ? does it need to be added here? https://github.com/FreeCAD/FreeCAD/blob/61f1310aa59c95a7c866ac619b8699932d94da46/src/Mod/PartDesign/App/FeatureHole.cpp#LL718C8-L718C8
or should we just hide the depth field when countersink is selected?

donovaly, 2023-03-09:
> The image shows that the depths are different, that for the second hole is wrong.

What is wrong? The depth must be different because you chose a different angle.
When you compare e.g. an M5 screw with 90° and 60° angle you see that the 60° has a longer head. This is clear because if you have e.g. an angle of just 1° the conic part gets enormously long.

donovaly, 2023-03-09:
> the code that generates the shape does not take the depth parameter into account

It does: increase the depth and the diameter increases. (except for a counterdrill of course)

adrianinsaval, 2023-03-09:
> What is wrong? The depth must be different because you chose a different angle.

They both show the same depth in the UI

> It does: increase the depth and the diameter increases. (except for a counterdrill of course)

I specifically show a line of code where this parameter is being ignored... the part of the code where the actual operation is made ignores this parameter, in the gui there seems to be something to increase the diameter as you increase the depth but it doesn't do this very well or doesn't cover all cases

donovaly, 2023-03-09:
> They both show the same depth in the UI

Because the depth is in both cases 2 mm.
Take the example file and set the depth to 0 mm for both holes. Then it should be clear. The depth is the distance the top of the screw head is below the surface.

> I specifically show a line of code where this parameter is being ignored... the part of the code where the actual operation is made ignores this parameter, in the gui there seems to be something to [...]

As I wrote in the forum, it is a bug that the change of the depth in the parameters does not recalculate the upper diameter. I will have a look because the calculation should be done in App, not the Gui part of the Hole feature.

adrianinsaval, 2023-03-09:
Ah that makes more sense

donovaly, 2023-03-10:
I am closing this bug because the hole feature works as expected. Meanwhile I added a tooltip to make this clear and also added a note to the Wiki.

closed 2023-03-10 by donovaly