FreeCAD/FreeCAD #8782: [Issue] [Techdraw] Inster Projection view crashes FC

volau, 2023-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.60h
  * Assembly3 0.11.4
  * Assembly4 0.12.5
  * dodo 1.0.0
  * Help 1.0.3
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Hi,
the function 'insert projection view' causes Freecad to crash in any case.
best regards

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-08:
Can you please provide:
* a file that quickly can demonstrate the crash
* a step-by-step explanation on how to trigger crash
* Start FC from the command line, trigger crash and paste the output

volau, 2023-03-08:
Hi,
see attched sample file TechDraw-Demo.FCStd. to force the crash, 
1) open file
2) switch to WB TechDraw
3) select page
4) click insert projection view
5) wait a few seconds for the crash

Started FC from command line, trigger the crash and write attached FreeCAD.log, No other output in command box.
best regards

[FreeCAD.zip](https://github.com/FreeCAD/FreeCAD/files/10925389/FreeCAD.zip)

Volker

WandererFan, 2023-03-09:
Creating a ProjectionGroup works on these versions (1 Win, 1 Linux):

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32214 (Git)
Build type: Release
Branch: master
Hash: 0a5e59adae70878e5d89d40e76169889ed4fe6fe
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32209 (Git)
Build type: Unknown
Branch: master
Hash: c8c91c4f8e5efcb515461427f72294a58a72523e
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * sheetmetal 0.2.59
  * FreeCAD_Electric
  * Assembly4 0.12.4
  * A2plus 0.4.56a
[/code]

WandererFan, 2023-03-09:
[image]

WandererFan, 2023-03-09:
@volau - are you using a local build or a weekly build or ???

volau, 2023-03-09:
Hi,
I’m using weekly build.

The same behaviour I see on two different machines. On both I have installed FC 0.20.2 and V0.21.dev 32110. On one machine I uninstalled  both installations including user configs. After reinstallation of only V0.21.d 32110 the problem is even the same. I will test the behaviour in the next days on my private machine with a different antivirus software.

Thank you and best regards

xtemp09, 2023-03-09:
I encountered Heisenbug when I had 

- Qt versions compiled with different options (like ICU, OpenSSL, different compilers etc)
- different Python versions

in PATH.

@volau, could you check PATH?

volau, 2023-03-10:
no entries in PATH for QT and Python.
I use the weekly build version an no local compilation

xtemp09, 2023-03-10:
@volau, what about gcc/clang/msvc/OpenSSL/ICU in PATH?

volau, 2023-03-10:
@xtemp09 no such entry

xtemp09, 2023-03-10:
I can reproduce the crash. I have downloaded [FreeCAD-0.21.0dev-32110-Installer.exe](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-32110-Installer.exe) from the weekly builds [page](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds).

---

##### gdb reports:
```
warning: Critical error detected c0000374
Thread 1 "CrBrowserMain" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffab2e0f123 in ntdll!RtlIsZeroMemory () from C:\Windows\SYSTEM32\ntdll.dll
```

##### lldb reports:
```
* thread #1, stop reason = Exception 0x80000003 encountered at address 0x7ffab2e0f122
    frame #0: 0x00007ffab2e0f123 ntdll.dll`RtlIsZeroMemory + 163
ntdll.dll`RtlIsZeroMemory:
->  0x7ffab2e0f123 <+163>: jmp    0x7ffab2e0f131            ; <+177>
    0x7ffab2e0f125 <+165>: movq   0xf8(%rsp), %rdi
    0x7ffab2e0f12d <+173>: movl   0x20(%rsp), %ebx
    0x7ffab2e0f131 <+177>: movl   %ebx, 0x30(%rsp)
```

---

I **cannot reproduce** the crash with any other version.

@volau, could you try newer weekly builds? The other versions are [here](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds).

xtemp09, 2023-03-10:
Here is the backtrace for the thread:

<details>
<summary>Backtrace from GDB</summary>

```c++
warning: Critical error detected c0000374

Thread 1 "CrBrowserMain" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffab2e0f123 in ntdll!RtlIsZeroMemory () from C:\Windows\SYSTEM32\ntdll.dll
(gdb) bt
#0  0x00007ffab2e0f123 in ntdll!RtlIsZeroMemory () from C:\Windows\SYSTEM32\ntdll.dll
#1  0x00007ffab2e17ee2 in ntdll!RtlpNtSetValueKey () from C:\Windows\SYSTEM32\ntdll.dll
#2  0x00007ffab2e181ca in ntdll!RtlpNtSetValueKey () from C:\Windows\SYSTEM32\ntdll.dll
#3  0x00007ffab2e1de51 in ntdll!RtlpNtSetValueKey () from C:\Windows\SYSTEM32\ntdll.dll
#4  0x00007ffab2d35bf0 in ntdll!RtlGetCurrentServiceSessionId () from C:\Windows\SYSTEM32\ntdll.dll
#5  0x00007ffab2d347b1 in ntdll!RtlFreeHeap () from C:\Windows\SYSTEM32\ntdll.dll
#6  0x00007ffab04cf05b in ucrtbase!_free_base () from C:\Windows\System32\ucrtbase.dll
#7  0x00007ffa68eb8230 in FreeCADGui!?getDisplayModes@ViewProvider@Gui@@UEBA?AV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@XZ
    () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#8  0x00007ffa42c83ab0 in TechDrawGui!?getDisplayModes@ViewProviderProjGroupItem@TechDrawGui@@UEBA?AV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@XZ () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDrawGui.pyd
#9  0x00007ffa68ec16d7 in FreeCADGui!?attach@ViewProviderDocumentObject@Gui@@UEAAXPEAVDocumentObject@App@@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#10 0x00007ffa42c86cda in TechDrawGui!?attach@ViewProviderDrawingView@TechDrawGui@@UEAAXPEAVDocumentObject@App@@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDrawGui.pyd
#11 0x00007ffa68bf8c9c in FreeCADGui!?slotNewObject@Document@Gui@@IEAAXAEBVDocumentObject@App@@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#12 0x00007ffa6a4036f4 in FreeCADApp!?create@Document@App@@SAPEAXXZ ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADApp.dll
#13 0x00007ffa6a3c5dc5 in FreeCADApp!??8DocumentT@App@@QEBA_NAEBV01@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADApp.dll
#14 0x00007ffa6a3ecf7c in FreeCADApp!??OCellAddress@App@@QEBA_NAEBU01@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADApp.dll
#15 0x00007ffa6a3f9e43 in FreeCADApp!?addObject@Document@App@@QEAAPEAVDocumentObject@2@PEBD0_N01@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADApp.dll
#16 0x00007ffa43568516 in TechDraw!?addProjection@DrawProjGroup@TechDraw@@QEAAPEAVDocumentObject@App@@PEBD@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDraw.pyd
#17 0x00007ffa435dcb5b in TechDraw!?addProjection@DrawProjGroupPy@TechDraw@@QEAAPEAU_object@@PEAU3@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDraw.pyd
#18 0x00007ffa435dc5cc in TechDraw!?staticCallback_addProjection@DrawProjGroupPy@TechDraw@@SAPEAU_object@@PEAU3@0@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDraw.pyd
#19 0x00007ffa64883d42 in python38!PyCFunction_Call () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#20 0x00007ffa64864dca in python38!_PyEval_EvalFrameDefault ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#21 0x00007ffa6485e618 in python38!_PyEval_EvalCodeWithName ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#22 0x00007ffa6487315b in python38!PyEval_EvalCodeEx ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#23 0x00007ffa648730b9 in python38!PyEval_EvalCode () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#24 0x00007ffa64872ac6 in python38!PyArena_New () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#25 0x00007ffa64872a55 in python38!PyArena_New () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#26 0x00007ffa6487274d in python38!PyRun_StringFlags ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\python38.dll
#27 0x00007ffa6b6ade71 in FreeCADBase!?runString@InterpreterSingleton@Base@@QEAA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEBD@Z () from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADBase.dll
#28 0x00007ffa68c452e2 in FreeCADGui!?_runCommand@Command@Gui@@SAXPEBDHW4DoCmd_Type@12@0@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#29 0x00007ffa68c44cac in FreeCADGui!?_doCommand@Command@Gui@@SAXPEBDHW4DoCmd_Type@12@0ZZ ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#30 0x00007ffa42b6547a in TechDrawGui!?printAll@MDIViewPage@TechDrawGui@@QEAAXXZ ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\lib\TechDrawGui.pyd
#31 0x00007ffa68c44f6a in FreeCADGui!?_invoke@Command@Gui@@AEAAXH_N@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#32 0x00007ffa68c49a0d in FreeCADGui!?invoke@Command@Gui@@QEAAXHW4TriggerSource@12@@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\FreeCADGui.dll
#33 0x00007ffa54a91d49 in Qt5Core!?qt_static_metacall@QObject@@CAXPEAV1@W4Call@QMetaObject@@HPEAPEAX@Z ()
   from E:\temp\FreeCAD-0.21.0dev-32110-Installer\bin\Qt5Core.dll
#34 0x00007ffa5fb85dcf in Qt5Widgets!?activate@QAction@@QEAAXW4ActionEvent@1@@Z ()
[... cut ...]
    frame #55: 0x00007ffa4d8aef2c Qt5Core.dll`int QEventLoop::exec(class QFlags<enum QEventLoop::ProcessEventsFlag>) + 444
    frame #56: 0x00007ffa4d8b1a94 Qt5Core.dll`int QCoreApplication::exec(void) + 340
    frame #57: 0x00007ffa4f095cab FreeCADGui.dll`void Gui::Application::runApplication(void) + 11563
    frame #58: 0x00007ff7d3f941c7 FreeCAD.exe
    frame #59: 0x00007ff7d3f954f7 FreeCAD.exe
    frame #60: 0x00007ff7d3f94aa6 FreeCAD.exe
    frame #61: 0x00007ffab29c7034 kernel32.dll`BaseThreadInitThunk + 20
    frame #62: 0x00007ffab2d62651 ntdll.dll`RtlUserThreadStart + 33
```
</details>

WandererFan, 2023-03-10:
A local build at the same commit works:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: weekly32110
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]

The current conda build also works:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32198 (Git)
Build type: Release
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.8.16, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]

donovaly, 2023-03-12:
I cannot reproduce this. I just created a new weekly build. Please try this out and report back.

WandererFan, 2023-03-14:
A local build using libpack 2.8 at the same commit works correctly.

xtemp09, 2023-03-14:
Is there a script to compile locally the bundle? I couldn't find it.

There are many scripts to compile locally toolchains, like clang, gcc, Qt etc. You just run it in MSYS2 shell/busybox/PowerShell and it downloads/compiles/installs. The scripts to compile FreeCAD are not publicly available if they even exist.

WandererFan, 2023-03-14:
I don't know how to build a libpack.  If you have a libpack, then to build FreeCAD I just open the CMakeLists.txt as a solution in Visual Studio (or from the "Open solution" button in CMake.

I only use windows in situations like this where there is a windows only bug.  @donovaly would be the guy to ask.

xtemp09, 2023-03-15:
I don't know Python, but simple search gave me:

---
[[Issue] freecad crash when open addon manager](https://github.com/FreeCAD/FreeCAD/issues/8874)

> This afternoon i can't reproduce the crash even when i refresh the local cache of addon manager.

---
[Python Crashes Exception code 0xc0000374 ntdll.dll](https://bugs.python.org/issue42335)

> The crashes occur sporadically.

Python 3.8 (3.8.5)

---

i.e. I'm pointing my finger at Python. Perhaps, update to newer version (3.8.16) should be done. If one looks through the [changelog](https://docs.python.org/release/3.8.16/whatsnew/changelog.html) from 3.8.10 to 3.8.16, memory leaks and use-after-free errors have been fixed, though I don't think it directly concerns the code used by FreeCAD.

luzpaz, 2023-04-09:
Is this still an unresolved issue?

WandererFan, 2023-04-09:
Couldn't reproduce it in the next build, and haven't heard any news, so I guess we can close it.

closed 2023-04-09 by WandererFan