FreeCAD/FreeCAD #8791: [Feature Request] PD: Adding ISO keyway tool (working like the hole one)

69Piero, 2023-03-09:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=72005

### Subproject(s) affected?

PartDesign

### Idea description

An useful engineering feature to add in PD is the keyway tool (working like the hole one).

i imagine a keyway tool that, once run, it:
1) asks you the circolar border (diameter is calculated by 3d model so also the key size according to ISO table)
2) asks you a reference face for the keyway
3) asks you the key length

now the tool automatically

4) creates a datum plane placed at diameter/2 - keyway depth according to given user's reference
5) generates a sketch with the constrained slot according to input
6) makes a through all pocket

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NXTler, 2023-08-26:
Based on this, we could also add DIN6888 (woodruff keyway) and DIN 6887 (wedge keyway)

grubertechnic, 2023-09-10:
I would definitely not add any keys this in FreeCAD, and i say that as mechanical engineer. This is a pure user specific and for that are macros or addons.

Besides that many many manufactures still does not use any standardization for that. Regularly i see some random keys with random dimensions (ok not all that random, some are calculated for the lowest production cost), not even a subspecies of a standardization.

pierreporte, 2024-08-25:
Does this feature already exist in a CAD program?

FEA-eng, 2024-08-26:
> Does this feature already exist in a CAD program?

It's typically just part of a shaft generator.

luzpaz, 2024-08-26:
> his should be moved to the FreeCAD-addons repository. So people don't have to bother about it when they don't use it.

Maybe @alexneufeld would be interested in creating and addon for this ? :sweat_smile:

alexneufeld, 2024-08-28:
To be honest, this sort of thing doesn't really even make sense as an addon module. Because the addon doesn't get bundled with the files you create using it, other people who open such files also need to install the addon. This isn't great UX. Therefore, addon modules should be reserved for functionality that's genuinely difficult to get with the base FreeCAD install. For simple stuff like this, it's better in the long run if you just model it with the normal PartDesign tools.

luzpaz, 2024-08-30:
Should we close ticket ?

pierreporte, 2024-08-30:
@luzpaz No, I think it is useful, especially if integrated to a shaft tool. It’s only P3 though. I’m downgrading it to “nice to have”.