FreeCAD/FreeCAD #8792: [Backup-Directory]

Manuelzwei, 2023-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * offline-documentation 1.0.0-alpha
[/code]
```

### Subproject(s) affected?

None

### Issue description

Wouldn't it be useful if we can define a path to a backup directory?

At this place under the point
"Anzahl der zuletzt benutzten Dateien"

 
[image]

or in this screen:

[image]

I think, in the project directory are projects, in the backup directory are the backups from the projects.  

I could also imagine an automatic:

E:\...\Project_xyz.FCStd
E:\...\Project_xyz___backups\Project_xyz.FCStd1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-09:
Related to #6045

FlachyJoe, 2023-03-10:
Clone of #6697

adrianinsaval, 2023-03-28:
this should be closed as a duplicate

closed 2023-03-28 by chennes