FreeCAD/FreeCAD #8800: [Issue] 'Unit system' setting is hard to find in preferences

PaddleStroke, 2023-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Last
```

### Subproject(s) affected?

Core

### Issue description

Currently to change the unit system, one has to go to preferences, then to the 'unit' tab, which is at the end of the list. So for new users it's hard to find it. 
It is also annoying to change manually for users that need to change between documents.

Suggestion : 
Find a more prominent place for this setting.
Enable document specific unit system on top of system default unit system.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

vernzimm, 2023-03-09:
Agree, and normally I switch decimal place at the same time. Would be great if the default could be set for that (maybe in current preference pane), and there was a drop-down somewhere accessible on main screen to swap unit profiles.

PaddleStroke, 2023-03-10:
I have 2 ideas to solve this : 
[image]
[image]

@vernzimm @sliptonic What do you think? I think I like idea B better because it's less clutter. And it's probably intuitive to click on that to try to change unit.

Syres916, 2023-03-10:
This was discussed on the forum months ago, I even put a patch together using the TuxbWb, see https://forum.freecad.org/viewtopic.php?p=632572#p632572 but yorik decided to make the change in the Draft Wb so if you have Draft in Autoload preferences it's already available in any workbench.

sliptonic, 2023-03-10:
The solution from Draft is problematic. First, it is changing the current active schema AND changing the user's saved preference but it isn't clear to the user that this is what is happening.  In other words, the change isn't just for the session.  

Second, the draft selector doesn't give the full list of schemas.  It shows a subset of units.  This means it's impossible to select the Metric Small Parts and CNC schema this way or the FEM schema.  
That means there are two different ways to change the default schema but they work differently.  

Idea B is more economical with space but less intuitive.  I would advocate for idea A.   In the future, if real estate is needed in the status bar, it could be revised to idea B and the user would already be trained to look there.  

If you did want to go with idea B, the tooltip could show the name of the schema on hover.

In my opinion, this should be coded in core so it's consistent across all workbenches and removed from Draft.

Are any other workbenches adding tools to the status bar?  I haven't seen that before. It's an interesting possibility.  We should have some workbench guidance/standards for when to use this though.

closed 2023-03-22 by PaddleStroke