FreeCAD/FreeCAD #8804: [Issue] Menu Do Not Work in Fullscreen Mode

1John419, 2023-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=631769&hilit=fullscreen#p631769

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

Report as fixed #7563.

However, on my machine, the problem still exists. Entering Fullscreen, either by View>Fullscreen or by Alt+F11, disables the menus. The only way to recover is via Alt+F11.

### Anything else?

Perhaps related: On Windows, if FC is closed with the window Maximized, FC loses its Restored window size and position.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-09:
It is fixed in 0.20.3 and 0.21dev which haven't been released yet and you are using 0.20.2 hence why the fault is still present in your build.

luzpaz, 2023-03-09:
Closing as duplicate

closed 2023-03-09 by luzpaz