FreeCAD/FreeCAD #8806: DXF import: "Group layers into blocks" doesn't work in weekly builds

alafr, 2023-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=665307#p665307

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * BIM
```

### Subproject(s) affected?

File formats

### Issue description

The option "Group layers into blocks" for DXF import does not work in weekly builds. It is probably a regression.
It does work in stable 0.20.2. This option creates a single Part object for all objects from a layer of the DXF file.

Screenshots are in the forum thread linked above.

The option "Get original colors from the DXF file" doesn't work (nor in weekly nor in stable 0.20.2). All imported objects are black. Ideally when "Group layers into blocks" and "Get original colors from the DXF file" are active it should set on each created Part the layer's color.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-03-13:
Both issues (group shapes + retrieve colors) confirmed. Looking into it...

alafr, 2023-03-13:
I searched the code history and although i have no C++ knowledge this commit looks suspicious for the broken "Group layers into blocks" option: https://github.com/FreeCAD/FreeCAD/commit/e1ebcef39e8f53ada00ad77ba800a0132aef61e8

```
>>> App.ParamGet("User parameter:BaseApp/Preferences/Mod/Draft").GetBool("groupLayers")
True
>>> App.ParamGet("User parameter:BaseApp/Preferences/Mod/Import").GetBool("groupLayers")
False
```

yorikvanhavre, 2023-03-15:
Yes it definitely looks like you found it. Now, we have two options: 

* Or we fix the Draft dialog at https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/Resources/ui/preferences-dxf.ui but there is more to fix in the whole Draft code )legacy importer) to make sure we go fetch the preferences values at the new location,
* Or maybe an easier option:
the readDXF function takes an `optionSource` argument, that allows us to give a different path. So maybe we just need to use that here: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/importDXF.py#L2826

I'll try the last one above when I have a moment to see if it works

alafr, 2023-03-17:
This indeed works in the python console:
```
import Import
os.chdir("C:/Users/antla/Desktop/")
Import.readDXF("test.dxf", App.ActiveDocument.Name, True, "User parameter:BaseApp/Preferences/Mod/Draft")
```

I see that a new issue with the same root cause was opened with more details and possible solutions: https://github.com/FreeCAD/FreeCAD/issues/8884

To avoid the risk of having multiple developers working simultaneously on the same issue I propose to keep this issue only for the "Get original colors from the DXF file" problem.

kpemartin, 2023-03-19:
I have a fix for this and also #8884. I started writing a draft PR for it but found the branch was contaminated with other changes (in particular, from another PR that I have still pending) so I have to clean up the mess before submitting the new PR.

kpemartin, 2023-03-19:
Oops, to clarify, #8959 only addresses the "Group layers into blocks" part of the original version of this issue, which isn't really part of this issue any more.

kpemartin, 2023-03-24:
Is this referring to the legacy importer or the C++ one?

The C++ one does not do anything with this option setting (option path Mod/Draft/dxfGetOriginalColors)  and, looking at the code, does not appear to do anything with colors. So if this is referring to the C++ importer it should just read that the "Get original colors from the DXF file" option does not work (nothing to do with grouping into blocks).

On the other hand, the legacy importer handles colors per the option when making individual objects, but (again, from reading the code) not when grouping the layers into single drawing objects.

I may be mistaken, but colors are a property of the ViewObject and thus cannot be applied individually to parts of a compound, though on the other hand it is certainly possible to individually color (e.g.) faces of a 3D shape.

alafr, 2023-03-24:
I was referring to the C++ importer. You are right, the colors are never imported, even without grouping into blocks.

In the legacy importer, colors are indeed imported for objects individually, not when grouped.

DXF layers have a color property and that color is used for entities whose individual color is "ByLayer".
When making a Freecad Compound from a DXF layer it would be great if the Freecad color matched the color of the DXF layer.
Obviously DXF entities whose color is not "ByLayer" will not have their correct color when grouped in a Compound (it is often considered as bad practice to set entity colors individually in Autocad anyway).

Here are the python commands that i was using after the import as a proof of concept (it doesn't work 100%, sometimes the freecad object has not exactly the same name as the layer...):

```
import ezdxf
doc = ezdxf.readfile("file.dxf")
for layer in doc.layers:
    rgb_color = ((0, 0, 0) if layer.color == 7 else ezdxf.colors.aci2rgb(layer.color))
    fc_color = tuple(a / 255 for a in rgb_color)
    # print(layer.dxf.name, layer.color, rgb_color, fc_color)
    obj = App.ActiveDocument.getObject(layer.dxf.name)
    if obj:
        obj.ViewObject.PointColor = fc_color
        obj.ViewObject.LineColor = fc_color
        obj.ViewObject.ShapeColor = fc_color
    else:
        print("Layer not found", layer.dxf.name)
```

luzpaz, 2023-04-16:
bumping for visibility

kpemartin, 2023-04-16:
This is on my personal list of things to do, I'll get to it after tax season is over (and I've had a chance to recover)

luzpaz, 2023-06-10:
@kpemartin :wave:

kpemartin, 2023-06-11:
Yep, I'm still here, deciding on a plan of attack. I think I want to start by refactoring the code so the common attributes like layer and colour (hence the relevance to this issue) are handled by a single piece of code for all drawing entity types. Right now this code exists in perhaps half a dozen places because it is handled separately for lines, polylines, ellipses, text, etc.

luzpaz, 2023-11-06:
bumping for status

kpemartin, 2023-11-20:
This actually looks harder than I expected because of the Gui/non-GUI issue.
To set the colour I need a View object for the drawing item, but the importing c++ code is in the `Import` project which does not have visibility to the C++ Gui classes. To do this the code would have to be in the `ImportGui` project.

`ImportGui` could have a derived class of `ImpExpDxfRead` that overrides code to assign colour (and perhaps other attributes like dashed lines). This would be in the `ImportGui` project, which would have its own python-callable `readDXF` function, and the Python code in Draft's `ImportDXF.open` could import `ImportGui` instead of `Import` if it knows the Gui is around, and call the extended `readDXF` function which creates the derived class of `ImpExpDxfRead` that knows how to assign colour.

Having said that, my grasp of the Gui/non-GUI division of FreeCAD is still tenuous, to say the least, so perhaps someone could tell me if I'm barking up the right tree with this approach.

yorikvanhavre, 2023-11-23:
It does not need to be very complicated, basically:

- When the GUI is not there, no color information is needed or used or stored. So it's a simple check, If the GUI is not there, we do nothing and this whole issue is not relevant. In Python we check `FreeCAD.GuiUp` to know if we have the GUI or not, in C++ I don't recall
- If the GUI is there, then we load the colors. However, there are a couple of different cases:
   - If we are loading object per object, then fine, we read the color (that is apparently [already implemented](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/dxf/dxf.cpp#L2897), and we apply it to the object. That is simply setting obj.ViewObject.LineColor (we still have to convert the autocad color to r,g,b though). Getting the layer color when the color is byLayer is apparently also already implemented...
   - If we are grouping objects by layer, then it's more complicated, as several objects become one, and currently objects don't support different line colors within a same object. I would do nothing for now until we figure out something

So we could maybe do this:

- Start by making m_ColorIndex accessible to `ImpExpDxfRead`, maybe the same way as  `LayerName()`, ideally turned into r,g,b values
- In either ImpExpDxfRead::AddObject or AddGraphics (maybe better), check if the GUI is available and we have an associated ViewObject for our object (the latter test is actually enough, since if we have no GUI there will be no ViewObject)
- If yes, and provided we didn't create a compound, we apply the color
- We also need special treatment for the Draft objects (dimensions and texts)

kpemartin, 2023-11-23:
Most of what you describe is the easy parts.

If grouping by layer the layer's colour can be applied to the grouped object(s), essentially as if they were all BYLAYER colour. Also, there are some objects (text, for instance) which I expect can't be grouped with geometric objects but should still be coloured.

My problems were all hiding in your third step. I was unable to do anything in C++ regarding View objects without including <Gui/...> which IIRC lead to linker errors. I'm assuming the Import project avoids using the Gui so it can be linked in non-Gui environments (which I assume include the command-line FreeCAD program), and there is a sibling ImportGui project for cases that want the Gui.

yorikvanhavre, 2023-11-24:
As a very simple and hacky workaround, we can do that in Python, like 
```python
obj = App.getDocument(docname).getObject(objName)
if obj and obj.ViewObject:
    obj.ViewObject.LineColor = (r,g,b)
```

PR #11701 (2023-12-14): Make C++ DXF importer set entity colors per DXF contents
Fixes #8806
Fixes #6924
This change:

1. Provides a hook for GUI-dependent code in the C++ importer (since color is a Gui concept) via a new class `ImpExpDxfReadGui` derived from `ImpExpDxfRead` and ultimately created by importDXF.py if the ImportGui module is available.
2. Sets the color on created FreeCAD objects based on the color specified in the DXF file, including handling BYLAYER color, using a new virtual methods which do nothing in `ImpExpDxfRead` but set color in `ImpExpDxfReadGui`.
3. Refactors the reader code in CDxfRead to r [... cut]
closed 2023-12-14 by yorikvanhavre