FreeCAD/FreeCAD #8808: QBackingStore::endPaint() called with active painter...

vernzimm, 2023-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=525171

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32198 (Git)
Build type: Release
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.8.16, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_Info
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Not sure exactly what happened, but program began to bog down and report was spammed with
"QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?"

[report out.txt](https://github.com/FreeCAD/FreeCAD/files/10937087/report.out.txt)

Note that forum link is to a similar sounding issue, but not related to Sketcher, and I couldn't figure out what the common feature was or how to reproduce it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-10:
It's a Qt bug, I don't know what causes it. Quick search showed that the message `QBackingStore::endPaint() called with active painter` introduced with [this patch](https://github.com/qt/qtbase/commit/2aa9908e24611fa6d321e694b8415ba7e8d364b0). `qt/qtbase 5.9.5 on Jan 12, 2018`

However, [it caused flooding](https://bugreports.qt.io/browse/QTBUG-66452) the console output with the aforementioned warning. `5.9.6, 5.10.1`
 
It was supposed to be fixed in [this](https://github.com/qt/qtdeclarative/commit/047f161b1a706e81034ccd8d4700a3126c16dfe4) patch. `5.11.0-beta1 on Feb 19, 2018`

Addition of `did you forget to destroy it or call QPainter::end() on it?` took place with [this](https://github.com/qt/qtbase/commit/86f91bf0f707e34392fbdf424d6914e7cf6bb4e6) patch. `qt/qtbase 5.12.5 on Jul 22, 2019`

Another bug with [similar warning](https://bugreports.qt.io/browse/QTBUG-83139) affecting `5.13.2, 5.14.1` on Linux, though, the bug [seemed to be unreproducible](https://bugreports.qt.io/browse/QTBUG-83139?focusedCommentId=506141&page=com.atlassian.jira.plugin.system.issuetabpanels:comment-tabpanel#comment-506141) with `5.14.2`

I hope that update to `Qt 5.15.8` will fix this.

vernzimm, 2023-03-10:
Thank you for the additional research. I regularly download the weekly updates, so I will check back in if there is anything additional.

xtemp09, 2023-03-11:
I meant just Qt. If I don't know what causes the flood, I can't point finger at anything. 😞

vernzimm, 2023-03-11:
@xtemp09 it's okay, thank you for your help. Is there something more I could do to get more information? Is there a way to turn on debug logging with more info or something?

xtemp09, 2023-03-12:
To my knowledge there is no debug logging.

Actually, if you want to help, you can check the latest build version. You used `0.21.0.32198`, while [the latest](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) is `0.21.0.32258`.

If the problem remains a thread backtrace would be good. [This is done](https://lldb.llvm.org/use/map.html) with _lldb_ or _gdb_.

xtemp09, 2024-09-15:
Is this issue still relevant?

vernzimm, 2024-09-15:
I going to say no, we can close it. I haven't had this happen in recent memory.

closed 2024-09-15 by vernzimm

AliT-Vaarst, 2024-10-01:
Still an issue as of PySide 6.7.2, think it occurs when the same box gets generated twice (i.e. user hasn't clicked the button in the box)

AliT-Vaarst, 2024-10-01:
Happens even if the box has been closed (with a QMessage.critical() box)