FreeCAD/FreeCAD #8814: [Problem] Sketcher: Element Widget. Checkboxes not clicking correctly.

PaddleStroke, 2023-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32237 (Git)
Build type: Release
Branch: master
Hash: d27e69bdc9cf42489067ec2fbda8e17a17d35929
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

The Element widget just got new checkboxes to toggle visibility. Same as it was implemented in constraint widget.
However the checkboxes don't work properly. Clicking on the checkboxes don't toggle them on the first try. It first select the row. Then you need to click again.
GeoPoints can't be toggled at all.
Sometimes clicking on one checkbox, toggle another checkbox (last selected it seems).

@abdullahtahiriyo I think you implemented those checkboxes last week?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

abdullahtahiriyo, 2023-03-10:
There is an issue with the selection of the row. I will need to take a look into it.

luzpaz, 2023-11-30:
@PaddleStroke is this still an issue ?

PaddleStroke, 2023-11-30:
Just tested and yes

abdullahtahiriyo, 2023-12-09:
The current implementation is not good. I would like to address this as part of the changes to the widgets using a proper model-view architecture. Hopefully during this development cycle.

PR #12054 (2024-01-23): Sketcher: Apply styling to elements and constraints UI
This ensures that checkboxes drawn in the Elements UI of the sketcher can be styled using the correct style.

Three changes are introduced:
1. paint() method uses style provided by the widget, not application.
2. Indicator is drawn using `PE_IndicatorCheckBox` not `CE_CheckBox` - it allows to use `::indicator` in qss to style it properly. This works the same as TreeView.
3. Fix for minor issues with text alignment and extension of selected items background to cover checkbox. This is how other controls of this kind work.

It also completely [... cut]
commit 2fac0122 (2024-01-23): Sketcher: Fix checkbox behavior for Element checkboxes

This creates new method overload for changing layer of exact
ElementItem. This fixes strange behavior of the checkboxes for elements
that were dependent not on what user clicked but on what is selected.

Fixes: #8814
closed 2024-01-23 by chennes