FreeCAD/FreeCAD #8815: [Issue] Workbench list is cluttered

PaddleStroke, 2023-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
last
```

### Subproject(s) affected?

Core

### Issue description

I heard a lot of people complaining that the workbench list is confusing / overwhelming for new users.

Little used workbenches are cluttering the list : <none>, Test Framework, Robot, Reverse Engineering, Raytracing, Points, OpenSCAD.

My suggestion is to disable them by default. People who need them are a small fraction of users and can easily add them.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-11:
I can provide you a long list of links to my posts/comments in which I complain on this topic. Many other users concur. Still, nothing has changed.. ='(

I would add, the removal is a strong move. I think it's better to add getting started wizard in the first run and allow the user to choose required workbenches.

Pesc0, 2023-03-11:
I agree with this. Also i think it would be a good idea to sort the list of active workbenches to have the most used ones on top.

Not really related (and perhaps not the right place to discuss this) but i think many toolbar elements could be hidden by default as well: print, cut, copy, paste, macro, many of the view buttons are already present in the navicube and right click menu... It would free up a lot of space, and if someone wants to have said elements they can always enable them.
IMO the top toolbar row should have the tools that are in common between workbenches, while the bottom row should contain only the workbench selector and workbench specific tools. 

In my experience having way too many buttons feels a bit overwhelming (too complicated) and thus unattractive to a new user. Plus a cleaner interface makes the software look more modern i think.

Also while we're at it antialiasing 2x should be enabled by default as it makes a massive improvement. I dont think there's an excuse for lower spec computers. 

I strongly support having a welcome wizard.

PaddleStroke, 2023-03-12:
Thanks for the ideas!

luzpaz, 2023-03-13:
related to #5772

chennes, 2023-03-13:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/8692

luzpaz, 2023-03-14:
Closing in favor of #8692

closed 2023-03-14 by luzpaz