FreeCAD/FreeCAD #8821: [Issue] Bug when creating substractive Helix

pinpox, 2023-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: NixOS 23.05 (Stoat)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Release
Python 3.10.10, Qt 5.15.8, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/Denmark (en_DK)
[/code]
```

### Subproject(s) affected?

None

### Issue description

I'm trying to create a substractive Helix in the Part Design Workbench. 
The preview looks fine:
[image]

But when creating it, it just breaks the part in a strange way:
[image]

I'm trying to build the substractive helix from a sketch that is on a datum plane. 
I have attached the project file here

### Anything else?

[soap_threaded.zip](https://github.com/FreeCAD/FreeCAD/files/10942711/soap_threaded.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-10:
Subtractive helix is getting love in the [dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) branch. Please test on development branch   
Also be sure to check:   
https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD+subtractive&type=issues&state=closed
https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD+subtractive&type=issues&state=open

pinpox, 2023-03-12:
I managed to work around the issue above, but have run into other issues with the substractive helix. I would like to try the `dev` branch, but the AppImage is throwing errors when I try to run it.

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pinpox/.cache/appimage-run/555b14f187ef6534afe33aff1cc61ad6065fe2a47aab954fe047a39bfb53d0c0/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: libpci.so.3: cannot open shared object file: No such file or directory
libpci.so.3: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pinpox/.cache/appimage-run/555b14f187ef6534afe33aff1cc61ad6065fe2a47aab954fe047a39bfb53d0c0/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'ImportError'>: libpci.so.3: cannot open shared object file: No such file or directory
libpci.so.3: cannot open shared object file: No such file or directory
```

Shouldn't the AppImage be self-contained and have that file in it?

luzpaz, 2023-03-13:
> `<class 'ImportError'>: libpci.so.3: cannot open shared object file: No such file or directory
libpci.so.3: cannot open shared object file: No such file or directory`

https://github.com/FreeCAD/FreeCAD-Bundle/issues/147#issuecomment-1366771948

luzpaz, 2023-03-15:
@pinpox can you please try the workaround referenced in the linked ticket and report back ?  
I'm closing this ticket for now as it seems this is a duplicate. If it's not a dupe, please re-open.

closed 2023-03-15 by luzpaz

pinpox, 2023-03-15:
> @pinpox can you please try the workaround referenced in the linked ticket and report back ? I'm closing this ticket for now as it seems this is a duplicate. If it's not a dupe, please re-open.

I'm not on ubuntu, I'm using NixOS and have not been able to find that package. But isn't the point of AppImage to be self-contained?

pinpox, 2023-03-16:
I got the dev version to work. The original substractive helix problem seems to be gone. Freecad crashed on me when working with substractive helix though, but I can't pinpoint it to a specific task.

I have noticed the performance of (substractive) helix is very poor though. Creating a substractive helix and doing a polar pattern with it (20 repetitions) took freecad about 3 minutes to compute, on a reasonably speced computer (i7 24GB Ram),  is that to be expected?