FreeCAD/FreeCAD #8827: The Helix function in both Part workbench and Part Design workbench will only create a Helix on the Z Axis

RoyKO1405, 2023-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.55
  * Help 1.0.3
  * ThreadProfile 1.84.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

in version 19 there was a special box to select the Axis on which you require the helix in this version it has disappeared. I have tried all the various boxes I can adjust its height and angle but no matter what I do I cannot set the Axis It only appears on the Z Axis and will not change.

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-21:
use the rotation axis and angle as desired to achieve or change the placement property after generating the helix. For example in your picture I would set rotation axis to x=1,y=0,z=0 and angle to -90deg.

I'm inclined to close this issue, thoughts?

adrianinsaval, 2023-04-21:
I checked now and the dialog is the same in 0.19 so really there is no regression here, closing.

closed 2023-04-21 by adrianinsaval