FreeCAD/FreeCAD #8839: Freecad SIGSEGV, Segmentation fault when edit face on cube

navr32, 2023-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Version freecad-git 0.21.0.32249.gda98cfc4d1-1 manjaro package.

[code]
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32249 (Git)
Build type: None
Branch: master
Hash: da98cfc4d1dab6e5f0f0fc6ca1f4f39cfdf1d729
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * sheetmetal 0.2.49
  * Plot 2022.4.17
  * FeedsAndSpeeds 0.4.0
  * LCInterlocking
  * fasteners 0.3.46
  * GDML 1.8.0
  * BCFPlugin 1.0.0
  * EM
  * BIM 2021.12.0
  * parts_library
  * workfeature
  * Curves 0.5.0
  * Design456 0.0.1
  * Trails 2022.1.0
  * POV-Ray-Rendering
  * kicadStepUpMod 10.13.2
  * CurvedShapes 1.0.3
  * Pyramids-and-Polyhedrons
  * fcgear 1.0.0
  * 3D_Printing_Tools
  * InventorLoader 1.3.0
  * 3DfindIT 1.2.0
  * Render 2022.2.0
  * BOLTSFC
  * Reinforcement
  * CubeMenu
  * OpticsWorkbench 1.0.8
  * Manipulator 1.4.5
  * lattice2 1.0.0
  * Glass
  * FEM_FrontISTR 0.1.0
  * WebTools
  * Beltrami 1.0.6-alpha
  * ThreadProfile 1.81.0
  * pyrate
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

When i try to edit a second face a cube create from extrude from a sketch.

Program received signal SIGSEGV, Segmentation fault.
[... 38 lines cut ...]
#38  0x7f6f3f1c005c in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x31c
#39  0x7f6f3f1c005c in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x31c
#40  0x7f6f3f1bec8e in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x28e
#41  /usr/lib/libQt5Widgets.so.5(+0x184aa5) [0x7f6f3f19faa5]
#42  0x7f6f3f1ca1e4 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x474
#43  0x7f6f3f193b5c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x9c
#44  0x7f6f40a091a9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#45  0x7f6f3e71af48 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#46  0x7f6f3e71ba53 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt5Core.so.5+0x193
#47  /usr/lib/libQt5Core.so.5(+0x2d4e88) [0x7f6f3e761e88]
#48  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x19b) [0x7f6f3c55582b]
#49  /usr/lib/libglib-2.0.so.0(+0xaccc9) [0x7f6f3c5accc9]
#50  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x32) [0x7f6f3c5540e2]
#51  0x7f6f3e765c6c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6c
#52  0x7f6f3e7136ec in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#53  0x7f6f3e71e219 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x99
#54  0x7f6f4093c232 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x1e92
#55  /usr/bin/FreeCAD(+0x85b9) [0x55d6451ff5b9]
#56  /usr/lib/libc.so.6(+0x23790) [0x7f6f3e071790]
#57  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f6f3e07184a]
#58  /usr/bin/FreeCAD(+0x8905) [0x55d6451ff905]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-11:
Please put output in code tags especially if there are is #<number> which will spam unrelated tickets. I've edited your ticket to reflect the change. TIA

luzpaz, 2023-03-11:
Please provide an example file for testers to quickly reproduce the bug. Also provide a clear step-by-step. TIA!

luzpaz, 2023-03-13:
@navr32 ping

luzpaz, 2023-03-19:
@navr32 please respond

adrianinsaval, 2023-03-28:
I suggest closing, this stinks of an AUR package that was not recompiled after a system upgrade.

The description is too vague to even attempt to reproduce and no example file is provided, if OP is sure his installation is actually working and can provide clear reproduction steps as well as a file, then they can reopen the issue.

adrianinsaval, 2023-03-29:
Please close the issue then, like I said this was most likely just a case of not rebuilding the AUR package after dependencies where updated.