FreeCAD/FreeCAD #8845: [Problem] FreeCAD uses outdated UI style by default

adrianinsaval, 2023-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Problem description

[image]

https://github.com/FreeCAD/FreeCAD/tree/master/src/Gui/QSint/actionpanel
https://github.com/FreeCAD/FreeCAD/blob/57aac275c77ecfd0295bc6ab78f77e57fbbbcda5/src/Gui/TaskView/TaskView.h#L115

https://marketplace.qt.io/products/qsint

> QSint::Actions module contains additional Qt widgets which maintain look-and-feel of Windows XP Task Panels, Windows Vista Actions etc.

Presumably this was done to give FreeCAD a more native look on windows, however nowadays we don't even support those windows versions anymore and it is having the opposite effect as MS has long abandoned that style of UI. This is also hindering new development: #8838 

It would be better to do away with the gradient and use a plain color as the other dock widgets.

This also negatively affects the linux version where a gradient is also added (although it at least uses colors derived from the system theme) even though the system theme does not use gradients for dock widgets, making it look out of place too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-11:
this is also somewhat related to #7623

xtemp09, 2023-03-11:
This topic was brought up many times, now even twice a day. 

QSint is actually one of a few things I like. XP-like interface is what I miss the most. =(

donovaly, 2023-03-11:
What is the actual bug? You might like the current default color or not but this is no bug.

adrianinsaval, 2023-03-11:
Not a bug but a problem as stated in the title, as explained this has the opposite effect of what it was supposed to do, it looks more out of place on windows than without it (you know people always say freecad looks old) and more importantly it is hindering development of a new feature.

> You might like the current default color or not but this is no bug.

My or your personal taste is not the point here but the expectation that most users would have and much more importantly the functionality that a plain background gives as we would not need frames for stuff that is expected to be always visible and fixed like the sketcher solver messages, allowing us to save space in the task panel.

> QSint is actually one of a few things I like. XP-like interface is what I miss the most. =(

you are most likely a minority though, most people expect a modern looking interface and are put off by xp era looking software.

MisterMakerNL, 2023-03-12:
I agree I kinda want a less extreme version of mt extreme pro dark to become default. And I also have a "classic" version that looks like this default version for the ones that rather want the old style. With all the newest bug fixes.

adrianinsaval, 2023-03-12:
as a more achievable goal instead of directly adopting a whole new stylesheet I would prefer to just update the stylesheet that we currently use for the task dialog to something that looks more cohesive with the rest of the interface on windows and that uses a readable background for the panel so that it won't look bad if we decide to put widgets directly on the panel.
The stylesheet is defined here:
https://github.com/FreeCAD/FreeCAD/blob/4cd350d2d0fb14cf8acea455ee656cead293272d/src/Gui/QSint/actionpanel/freecadscheme.cpp#L38-L112
https://github.com/FreeCAD/FreeCAD/blob/4cd350d2d0fb14cf8acea455ee656cead293272d/src/Gui/QSint/actionpanel/freecadscheme.cpp#L114-L132
https://github.com/FreeCAD/FreeCAD/blob/4cd350d2d0fb14cf8acea455ee656cead293272d/src/Gui/QSint/actionpanel/freecadscheme.cpp#L279-L347

I also notice that we are hardcoding left alignment of some text fields, which doesn't seem very friendly to users of a right aligned language.

adrianinsaval, 2023-03-12:
Can somebody explain what this whole QSint thing is providing that Qt itself doesn't? Would it be worth dropping this thing?

MisterMakerNL, 2023-03-12:
I tried patching up an older theme but it's too much work to get it up-to-date.
It was way easier to throw it out and start with a fresh default.

adrianinsaval, 2023-03-12:
Please have a look at the code I linked, these are not the .qss stylesheets that you find under sec/Gui/Stylesheets these way simpler stylesheets (defined inside a cpp file) applied only to some elements of the task panel

MisterMakerNL, 2024-07-22:
I think this can be closed due to having the startpage theme selection.

luzpaz, 2024-07-22:
@adrianinsaval what say you ?

krushia, 2024-08-16:
This is still an issue. I see no reason for a QSint widget to be used instead of a QToolbox. Not only is the header kinda ugly, but it wastes screen real estate. Also the panes can't be resized to fill the dock:

[image]

krushia, 2024-08-16:
Is it possible to separate what is currently called "Classic" into:
1. "Native" (uses defaults from operating system settings whenever possible. Unstyled Qt)
2. "Classic" (looks like "old FreeCAD") using OSint and other non-native things that @xtemp09 and others are accustomed to

maxwxyz, 2025-05-21:
@kadet1090 @MisterMakerNL is this still relevant?