FreeCAD/FreeCAD #8847: [Problem] Bundled stylesheets make sketcher task panel icon only toolbuttons with menu too small

adrianinsaval, 2023-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32220 (Git)
Build type: None
Branch: master
Hash: 088f51632b37279b74be1cfcc1a94779734a53bd
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods:
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Sketcher's constraint and element list widgets where reworked by paddle (a big improvement over the previous UI IMO), it looks very well when not using stylesheets:
[image]
but as soon as you enable one of the built-in stylesheets the icon only toolbuttons with menu are too narrow. ProDark as an example but the same happens with most or all of them:
[image]
the toolbuttons are also some secondary issues with the stylesheet:

- the toolbuttons are not as tall as the eye button
- the filters menu has a big ugly border around it

this seems to be where these toolbuttons are customized (similar lines can be found in all stylesheets):
https://github.com/FreeCAD/FreeCAD/blob/57aac275c77ecfd0295bc6ab78f77e57fbbbcda5/src/Gui/Stylesheets/Darker-blue.qss#L1585-L1598
they likely also need this hack applied for another toolbutton type in the stylesheet:
https://github.com/FreeCAD/FreeCAD/blob/57aac275c77ecfd0295bc6ab78f77e57fbbbcda5/src/Gui/Stylesheets/Darker-blue.qss#L1967-L1971
but I'm not sure how to apply this specifically to toolbuttons with a menu in the task panel as apparently toolbuttons are also used in other places without a menu (I wonder why those aren't pushbuttons?) and look odd with the 20px padding only on the right.

### Anything else?

previously discussed at https://forum.freecad.org/viewtopic.php?t=74118 and https://github.com/FreeCAD/FreeCAD/pull/8838#issuecomment-1465049454
and one of the panel reworks PR but can't find it at the moment, not much of relevance was said there anyways besides identifying that this is a bug in the bundled stylesheets only.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-12:
@MisterMakerNL @pgilfernandez any clues on this?

PaddleStroke, 2023-03-12:
I just saw that the eye button is taller than the other because it's a 'QPushButton'. So we could at least make them all the same size by changing it to QToolButton. To ensure consistency.

MisterMakerNL, 2023-03-12:
Don't see any issues in ExtremeProDark.
[image]

adrianinsaval, 2023-03-12:
> I just saw that the eye button is taller than the other because it's a 'QPushButton'. So we could at least make them all the same size by changing it to QToolButton. To ensure consistency. Can you please add this change to your PR?

notice that I made a PR to fix this already

> Don't see any issues in ExtremeProDark. Except the tiny fold out icon has a background, which it shouldn't. BTw it seems to be caused by using a toolbar icon instead of using a QSpinbox.

a spin box would not be correct for this, it is a qtoolbutton, you might need to add something specific to these toolbuttons for your theme, check out my PR, I fixed it for the bundled stylesheets

PaddleStroke, 2023-03-12:
Oh sorry I missed it.

PR #8849 (2023-03-14): fix #8847 sketcher dialog tool buttons bad styling
Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or changed o [... cut]
commit 0b2a2a8c (2023-03-14): fix #8847 sketcher dialog tool buttons bad styling
closed 2023-03-14 by abdullahtahiriyo