FreeCAD/FreeCAD #8848: [Feature Request] Align view to a Face

leoheck, 2023-03-12:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76717

### Subproject(s) affected?

Other (specify in description)

### Idea description

So, the idea here is to have a way to align/focus the view to a plane.

Similar to what we have for the standard views with the shortcuts 0, 1, 2, 3, 4... we could have a button and/or shortcut where we set the whole canvas view to align with the selected plane/face

Following the discussion on Freecad Forum, I noticed that this would be a nice addition to the Freecad.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jimgwhit, 2023-03-12:
This is already in Freecad:  Draft - Utilities - Select Plane

I use it all the time.

adrianinsaval, 2023-03-28:
@wwmayer do you know if it's possible to create a tool like this without calling Part API?

wwmayer, 2023-03-29:
This can be done with Coin3D API. If the user clicks on a face then Coin3D delivers a SoPickedPoint that contains information about the point on the surface and the normal.

However, the suggested workflow ignores the fact that such a rotation is not unique. And this may lead to unexpected results. If things should be unique then (at least optionally) an Up vector should be defined. This can be done by choosing a second point on the face.

maxwxyz, 2023-10-30:
I think I've created a somehow duplicate with https://github.com/FreeCAD/FreeCAD/issues/11234 (did not find this issue in GH search only mentioned in some forum posts), but I have also detailed more use cases.

closed 2024-02-17 by maxwxyz