FreeCAD/FreeCAD #8874: [Issue] freecad crash when open addon manager

fch77700, 2023-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.2
```

### Subproject(s) affected?

None

### Issue description

freecad 0.20.2 crash when open addon manager
OS : ubuntu 22.4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-13:
@fch77700 more info please. Can you run FreeCAD from the command line and see what it outputs when you recreate the crash? 
Also we need [Full About info](https://wiki.freecad.org/About). Please add it to the top post.

luzpaz, 2023-03-13:
Probably installed via Ubuntu repo or PPA ?  If so, can you recreate on official build ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20.2

fch77700, 2023-03-14:
This afternoon i can't reproduce the crash even when i refresh the local cache of addon manager. 
I

luzpaz, 2023-03-15:
@fch77700 please add full [About](https://wiki.freecad.org/About) info

luzpaz, 2023-03-15:
Closing ticket. If @fch77700 can reproduce, please reopen.

kinghat, 2023-06-11:
i can reproduce this. usually need a refresh of the cache. this is from appimage. it helps to either logout or reboot to start fresh.

> OS: Fedora Linux 38 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.0.33345 (Git) AppImage
> Build type: Release
> Branch: master
> Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
> Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
> Locale: English/United States (en_US)
> Installed mods: 
>   * Curves 0.6.10
>   * lattice2 1.0.0

```
[kinghat@kinghat-thinkpad Downloads]$ ./FreeCAD_weekly-builds-33345-2023-06-11-conda-Linux-x86_64-py310.AppImage 
FreeCAD 0.21.0, Libs: 0.21.0R33345 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Could not find the Qt platform plugin "wayland" in ""
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
git status failed for Curves
git status failed for lattice2
An error occurred updating macros from GitHub, trying clean checkout...:
Git returned a non-zero exit status: 128
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros

Returned stderr:
Cloning into '/home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros'...
fatal: unable to access 'https://github.com/FreeCAD/FreeCAD-macros.git/': Could not resolve host: github.com

/home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros
Attempting to do a clean checkout...
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAZcqsja/usr/Mod/AddonManager/addonmanager_git.py", line 352, in _synchronous_call_git
    proc = utils.run_interruptable_subprocess(final_args)
  File "/tmp/.mount_FreeCAZcqsja/usr/Mod/AddonManager/addonmanager_utilities.py", line 446, in run_interruptable_subprocess
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['/usr/bin/git', 'clone', '--recurse-submodules', 'https://github.com/FreeCAD/FreeCAD-macros.git', '/home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros']' returned non-zero exit status 128.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/tmp/.mount_FreeCAZcqsja/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 327, in _update_local_git_repo
    self.git_manager.clone(
  File "/tmp/.mount_FreeCAZcqsja/usr/Mod/AddonManager/addonmanager_git.py", line 66, in clone
    self._synchronous_call_git(final_args)
  File "/tmp/.mount_FreeCAZcqsja/usr/Mod/AddonManager/addonmanager_git.py", line 354, in _synchronous_call_git
    raise GitFailed(
addonmanager_git.GitFailed: Git returned a non-zero exit status: 128
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros

Returned stderr:
Cloning into '/home/kinghat/.cache/FreeCAD/Cache/AddonManager/Macros'...
fatal: unable to access 'https://github.com/FreeCAD/FreeCAD-macros.git/': Could not resolve host: github.com

During handling of the above exception, another exception occurred:

[... 17 lines cut ...]
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Could not find the Qt platform plugin "wayland" in ""
Cannot find icon: /home/kinghat/.cache/FreeCAD/Cache/AddonManager/MacroIcons/Macro_cross_section.png
Cannot find icon: /home/kinghat/.cache/FreeCAD/Cache/AddonManager/MacroIcons/Macro_Align_View_to_Face.png
Cannot find icon: /home/kinghat/.cache/FreeCAD/Cache/AddonManager/MacroIcons/Easy-alias-icon.png
Cannot find icon: /home/kinghat/.cache/FreeCAD/Cache/AddonManager/MacroIcons/Aliasmanager_icon.png
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
```

lmk if i can provide any further details. the initial load of the addon manager seems flaky. if it doesnt crash freecad, sometimes it will just hang on updating something specific.

luzpaz, 2023-10-13:
@chennes does the error message above point to anything specific ?
It may start from:
> Could not find the Qt platform plugin "wayland" in ""

Thoughts? 

@kinghat is this still reproducible on 0.21.1 or 0.22dev ?

kinghat, 2023-10-13:
i cannot reproduce in either the appimage or flatpak.

luzpaz, 2023-10-15:
@kinghat Thanks, can you add about info of the versions you tested with please?

kinghat, 2023-10-15:
> @kinghat Thanks, can you add about info of the versions you tested with please?

yep sorry about that!

> OS: KDE Flatpak runtime (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.1.33694 (Git)
> Build type: Release
> Branch: (HEAD detached at f670854)
> Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
> Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
> Locale: English/United States (en_US)

> OS: Fedora Linux 38 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.34651 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: aa5bea4e2e8089bf00f61075a0860dacda42dc8b
> Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
> Locale: English/United States (en_US)

luzpaz, 2023-11-08:
CC @chennes

luzpaz, 2023-12-17:
Is this issue still relevant ? @kinghat & @fch77700 (please test latest dev version and report full version info)

kinghat, 2023-12-17:
> Is this issue still relevant ? @kinghat & @fch77700 (please test latest dev version and report full version info)

im actually not having this issue on the flatpak stable:

> OS: KDE Flatpak runtime (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.2.33771 (Git)
> Build type: Release
> Branch: (HEAD detached at b9bfa5c)
> Hash: b9bfa5c5507506e4515816414cd27f4851d00489
> Python 3.10.13, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
> Locale: English/United States (en_US)
> Installed mods: 
>   * BIM 2021.12.0
>   * ThreadProfile 1.89.0
>   * fasteners 0.4.75
>   * ExtremeProDark 2.9.0
>   * Manipulator 1.5.4
>   * OpenDark 2023.12.1

or on the dev afaict. but i dont use the dev version as daily driver anymore:

```
OS: Fedora Linux 39 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 (Git) AppImage
Build type: Release
Branch: main
Hash: ed10fe216d332debfd317d4639e284c50f5dab47
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

closed 2024-01-30 by chennes

crasu, 2024-02-02:
This issue happens to me as well on:

[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]