FreeCAD/FreeCAD #8875: [Issue] error while saving Path postporcessing

Krandelbord, 2023-03-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (X-Cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

None

### Issue description

When I click Post Process Selected job, and I type file name then I get the following exception in console:

post: KineticNCBeamicon2(/home/emil/eeee.gcode, )
18:57:06  Show editor = 1
18:57:06  postprocessing...
18:57:06  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAFPF5WR/usr/Mod/Path/PathScripts/PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "/tmp/.mount_FreeCAFPF5WR/usr/Mod/Path/PathScripts/PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "/tmp/.mount_FreeCAFPF5WR/usr/Mod/Path/PathScripts/PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "/tmp/.mount_FreeCAFPF5WR/usr/Mod/Path/PathScripts/post/KineticNCBeamicon2_post.py", line 227, in export
    job = PathUtils.findParentJob(obj)
  File "/tmp/.mount_FreeCAFPF5WR/usr/Mod/Path/PathScripts/PathUtils.py", line 405, in findParentJob
    for i in obj.InList:

'_TempObject' object has no attribute 'InList'

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-13:
Looks like a dupe of https://github.com/FreeCAD/FreeCAD/issues/8334#issuecomment-1418261575

derlui, 2023-07-27:
I am facing the exact same issue using the KineticNCBeamicon2 post-processor, some other post-processors work fine.

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 1397818c9664d158281339a84be59d3e9887f6ad
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.0
  * A2plus 0.4.60k
  * fasteners 0.4.56
  * Assembly4 0.12.7
  * kicadStepUpMod 10.18.6
[/code]

closed 2025-01-15 by github-actions[bot]